ros2/launch_ros #2: [launch_ros] launch_ros node should have an anonymous name

mikaelarguedas, 2019-03-22:
Currently if 2 launch files are launched using `launch_ros`, 2 nodes named `launch_ros` are created.

The nodes should have different names. 
In the other `ros2` tools it is currently done by using the process ID https://github.com/ros2/ros2cli/blob/4059a2f84addbbaab121b7f92b872dd5acffa4b3/ros2cli/ros2cli/node/direct.py#L36-L37'

MengNan-Li, 2019-05-27:
I have the same question. Starting multiple launch files in ROS2 will launch multiple '/launch ros' nodes, which will cause an error when executing the "ros2 param list" command.

MengNan-Li, 2019-05-27:
ros2 param list

Traceback (most recent call last):
  File "/opt/ros/crystal/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.6.3', 'console_scripts', 'ros2')()
  File "/opt/ros/crystal/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/crystal/lib/python3.6/site-packages/ros2param/command/param.py", line 40, in main
    return extension.main(args=args)
  File "/opt/ros/crystal/lib/python3.6/site-packages/ros2param/verb/list.py", line 80, in main
    rclpy.spin_once(node, timeout_sec=1.0)
  File "/opt/ros/crystal/lib/python3.6/site-packages/rclpy/__init__.py", line 98, in spin_once
    executor.spin_once(timeout_sec=timeout_sec)
  File "/opt/ros/crystal/lib/python3.6/site-packages/rclpy/executors.py", line 572, in spin_once
    raise handler.exception()
  File "/opt/ros/crystal/lib/python3.6/site-packages/rclpy/task.py", line 206, in __call__
    self._handler.send(None)
  File "/opt/ros/crystal/lib/python3.6/site-packages/rclpy/executors.py", line 324, in handler
    await call_coroutine(entity, arg)
  File "/opt/ros/crystal/lib/python3.6/site-packages/rclpy/executors.py", line 264, in _execute_client
    future = client._pending_requests[sequence]
KeyError: 1

dirk-thomas, 2019-05-29:
> `ros2 param list`

This should be fixed in the upcoming Dashing release.

PR #98 (2019-11-28): Add pid to launch_ros node name as suffix
Fixes #2 . As the title says, it adds the pid to the node's name as a suffix. Is there any need to create a test for this?
closed 2019-11-28 by BMarchi