ros2/launch_ros #6: Launch composable C++ nodes

sloretz, 2019-03-22:
## Feature request

This is a feature request for launching multiple composable nodes into the same process. The acceptance criteria for this ticket is limited to launching dynamically composable rclcpp non-lifecycle nodes into a container process.

#### Feature description

`launch_ros` must pass arguments to the composable nodes where "arguments" means:

* the information that can be passed via the command line of any `rcl` based node
  * parameters
  * remap rules
  * node name
  * namespace
  * log level
* information that is specific to the client library `rclcpp`
  * use_intra_process_comms
  * start_parameter_services

An exception is `use_global_arguments` which should always be `false` so a composed node is not affected by arguments passed to the container process.

Arguments does not include environment variables like `RMW_IMPLEMENTATION` which apply to the whole container process.

PRs
- [x] OnProcessStart ros2/launch#171
- [x] Reusable type checking for remap rules ros2/launch#173
- [x] Reusable type checking for parameters ros2/launch#192
- [ ] `launch_ros.ComposableNode`, `launch_ros.actions.LoadComposableNodes`, `launch_ros.actions.ComposableNodeContainer` plus examples from ros2/launch#168

#### Implementation considerations

##### launch_ros

See slide 8 of https://roscon.ros.org/2018/presentations/ROSCon2018_launch.pdf

```python
        ExecuteComposableNodeProcess(
            package='rclcpp_components', container_executable='rclcpp_node_container', output='screen',
            composable_node_descriptions=[
                ComposableNodeDescription(
                    package_name='demo_nodes_cpp', node_plugin_name='talker',
                    name='my_talker'),
                ComposableNodeDescription(
                    package_name='demo_nodes_cpp', node_plugin_name='listener'
                    name='my_listener')
            ]
        )
```

`ExecuteComposableNodeProcess` should start a container executable with a defined API. This allows a user to create a custom container process (e.g. create a container with a custom executor). The API may be a config file and/or a ROS API. The acceptance criteria for this ticket is limited to a ROS service API that can load and unload composable nodes (see also ros2/rcl_interfaces#60).

##### rclcpp

There must be a new ROS package (`rclcpp_components`?) with a container process for rclcpp nodes. The acceptance criteria for this task is to be able to use either a single threaded executor or multithreaded executor in a container process. For example, this could be two container processes, or one with a CLI argument to choose the executor.

Launching a composable node requires a common API for constructing a node.
One option is to mimic ROS 1 nodelets which required a default constructor on the node and had an `init()` member function to initialize the nodes with parameters.
Another option is having a constructor with one argument and and a factory class to call it.
Regardless this task should require ros2/rclcpp#492 so there is one object `NodeArguments/NodeOptions` so new arguments to a node don't require changing the signature of a function.

* https://github.com/ros2/rclcpp/tree/wjwwood/components_wip
* https://github.com/ros2/rclcpp/blob/wjwwood/components_wip/rclcpp_components/doc/source/architecture.rst
* https://github.com/ros2/rcl_interfaces/tree/wjwwood/components_wip

##### CMake API

This ticket includes creating a CMake API for creating a composable node. This API should

* call add_library(... SHARED ..)
* optionally add_executable() with a generated `main()`
* register the composable node with pluginlib

hidmic, 2019-04-30:
@sloretz @mjcarroll can we close this one?

mjcarroll, 2019-04-30:
Yup, I believe so.

closed 2019-04-30 by sloretz