ros2/launch_ros #7: [launch_ros] node_name overrides for a multi-node rclpy process

stonier, 2019-03-22:
#### Feature description

A means of setting `node_name` overrides for a multi-node rclpy process such as that given in #204.

#### Implementation considerations

This also brings into question the naming of the class `launch_ros.actions.Node` since it currently can launch a multi-node process. Perhaps that should fail on trying to launch a multi-node process and for that, use a `launch_ros.actions.MultiNode` action.

Note: This is a different problem to the composable node launch. Here the nodes are brought into the same process via ... the process,  not the launcher. I suspect the same situation would arise linking two c++ classes with nodes and instantiating them in the same application.

masynthetic, 2021-05-19:
is there a workaround/fix for this? remapping the `rclpy.create_node()` parameter does not seem to do anything, any use of the `name=` parameter in the launch file seems to cause the issue

hidmic, 2021-05-19:
> any use of the name= parameter in the launch file seems to cause the issue

@masynthetic  `launch_ros.actions.Node` is not a multi-node aware action, so yes, it won't work as you intend in that case.

If you can't use components, a possible workaround is to launch your process as a plain process using `launch.actions.ExecuteProcess` and configure your nodes through command line arguments.

Alternatively, contributions are most welcome. A multi-node aware action proposal would be great.

jbohren, 2021-12-02:
> This also brings into question the naming of the class `launch_ros.actions.Node` since it currently can launch a multi-node process.
> @stonier

This is what brought me here.

Since even a single-node executable is an _executable which owns a node_ and not the _node itself_, it feels like `launch_ros.actions.Node` is a misnomer and could lead to confusion and unexpected behavior. Something like `ExecuteSingleNodeProcess` would definitely reduce the likelihood of that occurring.

Also is it weird that some `actions` are written as imperative statements while others are noun phrases?