ros2/launch_ros #13: Node action failing with 'env' argument

ivanpauno, 2019-04-11:
## Bug report

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From source
- Version or commit hash:
  - launch_ros(#448846d9361f39bdde94d1a41679abac3a1e124a)
- DDS implementation:
  - fast-rtps
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
This is a modified version of `launch_ros/examples/pub_sub_launch.py` [here](https://github.com/ros2/launch_ros/blob/master/launch_ros/examples/pub_sub_launch.py).

```python3
# Copyright 2018 Open Source Robotics Foundation, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Launch a talker and a listener."""

import os
import sys
sys.path.insert(0, os.path.join(os.path.dirname(__file__), '..'))  # noqa
sys.path.insert(0, os.path.join(os.path.dirname(__file__), '..', '..', 'launch'))  # noqa

from launch import LaunchDescription
from launch import LaunchIntrospector
from launch import LaunchService

from launch_ros import get_default_launch_description
import launch_ros.actions

def main(argv=sys.argv[1:]):
    """Main."""
    ld = LaunchDescription([
        launch_ros.actions.Node(
            package='demo_nodes_cpp', node_executable='talker', output='screen',
            env={'asd': '1'},
            remappings=[('chatter', 'my_chatter')]),
        launch_ros.actions.Node(
            package='demo_nodes_cpp', node_executable='listener', output='screen',
            remappings=[('chatter', 'my_chatter')]),
    ])

    print('Starting introspection of launch description...')
    print('')

    print(LaunchIntrospector().format_launch_description(ld))

    print('')
    print('Starting launch of launch description...')
    print('')

    # ls = LaunchService(debug=True)
    ls = LaunchService()
    ls.include_launch_description(get_default_launch_description())
    ls.include_launch_description(ld)
    return ls.run()

if __name__ == '__main__':
    main()
```
This line has been added to the original file:

```python3
            env={'asd': '1'},
```

#### Expected behavior
It should work as the original example.

#### Actual behavior
It fails to find the rmw implementation:

```
Starting introspection of launch description...

<launch.launch_description.LaunchDescription object at 0x7fcf4cfff3c8>
├── ExecuteProcess(cmd=[ExecInPkg(pkg='demo_nodes_cpp', exec='talker'), LocalVar('remapping 1')], cwd=None, env={'asd': '1'}, shell=False)
└── ExecuteProcess(cmd=[ExecInPkg(pkg='demo_nodes_cpp', exec='listener'), LocalVar('remapping 1')], cwd=None, env=None, shell=False)

Starting launch of launch description...

[INFO] [launch]: All log files can be found below /home/ivanpauno/.ros/log/2019-04-11-16-24-06-965421-71c1584df2a2-5104
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [5113]
[INFO] [listener-2]: process started with pid [5114]
[talker-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[talker-1]   what():  failed to initialized rcl init options: failed to find shared library of rmw implementation. Searched rmw_fastrtps_cpp, at /home/ivanpauno/ros2_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:129, at /home/ivanpauno/ros2_ws/src/ros2/rcl/rcl/src/rcl/init_options.c:55
[ERROR] [talker-1]: process has died [pid 5113, exit code -6, cmd '/home/ivanpauno/ros2_ws/install/demo_nodes_cpp/lib/demo_nodes_cpp/talker chatter:=my_chatter'].
```

wjwwood, 2019-04-11:
It overrides the environment, it doesn't append to it. You're removed a needed environment variable by doing it this way. Instead you'd way to first copy `os.environ`, then add your environment variable, and then pass that to execute node.

closed 2019-04-11 by ivanpauno

wjwwood, 2019-04-11:
I will say, this is confusing default behavior, so maybe we should change the default to override specific keys, but are "updated" into a copy of the `os.environ` by default. That would avoid this pitfall, and we could still have an "override existing environment variables" option to enable the current behavior when someone actually wants that.

dirk-thomas, 2019-04-11:
The current API mimics the Python API of similar functionality (e.g. the `subprocess` API). I would rather stick to one consistent API than using different behavior and/or multiple options.

wjwwood, 2019-04-11:
Yes, that's why it is the way it is now, but on the other hand, if you're extending the environment from an XML file it might not be so convenient express that you want to copy the existing environment, update it, and then pass that.

ivanpauno, 2019-04-12:
> Yes, that's why it is the way it is now, but on the other hand, if you're extending the environment from an XML file it might not be so convenient express that you want to copy the existing environment, update it, and then pass that.

I exactly ran into this problem when working in the XML front-end. Is not natural to clear all the environment when you add a `<env>` tag. Although this could be worked around in the XML parsing, I prefer having a consistent behavior between the xml-frontend and the programmatic launch system.

If you also agree, I will reopen and solve it.

dirk-thomas, 2019-04-12:
I would argue that our Python API should follow the style of other Python API to match behavior. Everything else is unexpected.

An `env` tag in the XML should be implemented to extend/update the existing environment which I would think is the natural expectation for that interface.

wjwwood, 2019-04-13:
What about a different name? Then it's not overriding what other Python API's solve? Perhaps `additional_env` or something similar?

> I would argue that our Python API should follow the style of other Python API to match behavior. Everything else is unexpected.

Well, right now the behavior of env in our API is unexpected for our users, and I don't blame them, I think it's counter intuitive as well.

So I agree in general that we should try to match the Python API where it makes sense, but in this case I don't know that the Python API is such a good example to follow.

dirk-thomas, 2019-04-13:
> Perhaps `additional_env` or something similar?

Sounds good to me.

ivanpauno, 2019-04-15:
> Perhaps `additional_env` or something similar?

Sounds good, I will apply this change to both `Node` and `ExecuteProcess` classes.
Also I will add an additional bool argument `new_env`, which allows starting from a clean environment.

dirk-thomas, 2019-04-15:
> I will add an additional bool argument `new_env`, which allows starting from a clean environment.

I would rather keep `env` as is (a full environment) and add a new keyword argument `additional_env` which can will be used to update `env` (or the current content of `os.environ` if `env` isn't provided explicitly).

wjwwood, 2019-04-16:
I think `new_env` and `additional_env` have the benefit of forcing the user to go "wait, what?" and read the difference, but I'm also ok with `env` and `additional_env`, working together as @dirk-thomas described.

So @ivanpauno please do as @dirk-thomas suggested unless he changes his mind.

dirk-thomas, 2019-04-16:
> I think `new_env` and `additional_env` have the benefit of forcing the user to go "wait, what?" and read the difference ...

Sounds good too