ros2/launch_ros #19: Standalone Component Action

mjcarroll, 2019-04-23:
## Feature request

#### Feature description

Add a launch action that will allow for a single component to be loaded into a component container.  This would be a simplified representation or alias to the larger component actions.

hidmic, 2019-04-24:
So, currently one can do:

```python
ld = LaunchDescription([
    launch_ros.actions.ComposableNodeContainer(
        package='rclcpp_components', node_executable='component_container',
        composable_node_descriptions=[
             launch_ros.descriptions.ComposableNode(
                 package='composition', node_plugin='composition::Talker'
             )
        ]
    )
])
```

 @dirk-thomas @mjcarroll do you guys think that's too verbose? How do you imagine it could be? I wonder about the potential redundancy with the `Node` action once every ROS2 node becomes a composable node (which is where we're going if I'm not mistaken).

sloretz, 2019-04-24:
IIUC correctly the desire is for a standalone executable to be generated for components, something created from a template that has a `main()` and instantiates the component class. If so, that would remove the need for a standalone component action because `launch_ros.actions.Node` could be used instead.

jacobperron, 2019-12-12:
On a related note, I was a little confused by the `ComposableNodeContainer` API thinking I could do something like:

```python
ld = launch.LaunchDescription([
    launch_ros.actions.ComposableNodeContainer(
            node_name='my_container_name', node_namespace='/',
            composable_node_descriptions=[
                launch_ros.descriptions.ComposableNode(
                    package='composition', node_plugin='composition::Talker'
             )
        ]
    )
])
```

But I was missing the keyword arguments:

```python
    package='rclcpp_components', node_executable='component_container',
```

Do you think we could make these the default values for the `ComposableNodeContainer` action? I suppose this adds a runtime dependency on `rclcpp_components`. I think this would be enough to resolve this feature request.

As an aside, we could probably also relax the requirement for specifying the node name and namespace (similar to the `Node` action).

sloretz, 2019-12-12:
> Do you think we could make these the default values for the ComposableNodeContainer action? I suppose this adds a runtime dependency on rclcpp_components.

The API for component managers is language neutral. It is possible to implement a python component container for `rclpy` nodes. If `rclcpp_components` become the default, those values would still have to be set manually for other client libraries. Besides adding a runtime dependency, I'm concerned that could be perceived as  a preference for one client library over all others.

emersonknapp, 2021-03-05:
> IIUC correctly the desire is for a standalone executable to be generated for components, something created from a template that has a `main()` and instantiates the component class. If so, that would [...]

Hey - I was just playing with components and wanting this exact thing, mostly for interactive purposes - e.g. `ros2 run my_package my_component` with Ctrl+C to exit the node - without having to do 2 commands to start a container then load the component into it. Did this part of the functionality get built? Or, is there a dedicated issue tracking it?

jacobperron, 2021-03-08:
@emersonknapp We added some CMake magic to help build standalone executables from node components (https://github.com/ros2/rclcpp/pull/784). [For example](https://github.com/ros2/demos/blob/ed0299dc808f94b3c995dd4530680560e07226ff/demo_nodes_cpp/CMakeLists.txt#L81-L83), `demo_nodes_cpp` takes advantage of it:

```cmake
rclcpp_components_register_node(timers_library
  PLUGIN "demo_nodes_cpp::OneOffTimerNode"
  EXECUTABLE one_off_timer)
```

Basically, when registering the component with `rclcpp_components_register_node`, you can provide an optional "EXECUTABLE" name. If an executable name is given, then a standalone executable will also be built with that name. Then you can run the following for example:

```
ros2 run demo_nodes_cpp one_off_timer
```

So, if you use this CMake helper for your component, then you can launch it as a regular node (ie. with the `Node` action).

---

I'm still not exactly sure what the objective of this ticket is. I guess it would be to make the current syntax for loading a single component into a container less verbose. E.g.

```diff
- launch_ros.actions.ComposableNodeContainer(
-     package='rclcpp_components', executable='component_container',
-     composable_node_descriptions=[
-          launch_ros.descriptions.ComposableNode(
-              package='composition', plugin='composition::Talker'
-          )
-     ]
- )
+ launch_ros.action.StandaloneComponent(
+     package='composition', plugin='composition::Talker'
+ )

Though, to achieve this we would have to assume the type of container and declare a dependency on `rclcpp_components` or equivalent. Also, it may be possible to have Python components at some point and it's not clear to me how handle this.

Maybe we can close this ticket as something we won't do?

jacobperron, 2021-03-08:
Considering the extensible nature of launch, maybe it makes more sense to create a specialized package, `launch_rclcpp_components`, that provides this specific functionality using the container provided by `rclcpp_components`.

emersonknapp, 2021-03-08:
@jacobperron thanks for that extra context, the CMake stuff meets my needs - sorry to have derailed the thread slightly off topic.

My understanding of this issue is, yes, to just make that syntax less verbose. It probably doesn't need to exist in `launch_ros` - somebody could write it as a simple function in their project if it's a common use case (in which case though, why use components?)

> Also, it may be possible to have Python components at some point and it's not clear to me how handle this.

This seems possible by [embedding the python interpreter](https://docs.python.org/3/extending/embedding.html) - the component syntax would probably need to be a little different given that no special consideration needs to be made in the Python code, just a defined `Node` subclass - could look like `ComposableNode(package='my_python_pkg', python_plugin='my_python_pkg.nodes.ComponentNode')` - but that's definitely off topic for this ticket :)