ros2/launch_ros #34: Trying to launch a lifecycle node with launch_ros.actions.Node fails silently

rotu, 2019-06-10:
Passing a node executable to a `launch_ros.actions.Node` fails rather silently as in the below example. The mistake is hard to spot and should throw an obvious error pointing out the mistake (in this case that node_executable lifecycle_talker does not spin a lifecycle node).

```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(package='lifecycle', node_executable='lifecycle_talker',
             node_name='lc_talker', output='screen'),
        Node(package='lifecycle', node_executable='lifecycle_listener', output='screen'),
        Node(package='lifecycle', node_executable='lifecycle_service_client', output='screen')
    ])
```

> [INFO] [lifecycle_talker-1]: process started with pid [30809]
> [INFO] [lifecycle_listener-2]: process started with pid [30810]
> [INFO] [lifecycle_service_client-3]: process started with pid [30811]
> [lifecycle_service_client-3] [ERROR] [lc_client]: Service /lc_talker/change_state is not available.
> [INFO] [lifecycle_service_client-3]: process has finished cleanly [pid 30811]

ivanpauno, 2019-06-11:
I can't reproduce the issue, it works well for me.
Could you explain it further?
Also, add system details (Installed from source or binaries, etc).

rotu, 2019-06-11:
I'm on master branch. Closing this ticket as user error - it was not clear to me whether you could use `Node` action to launch a `LifecycleNode`. Turns out it works just fine, though I'm not sure whether this is by design or by accident.

closed 2019-06-11 by rotu

ivanpauno, 2019-06-11:
It is be design. `LifecycleNode` action does the same that `Node` action, and also emits some events that can be handled by user added event handlers in the launch file. If you are not using that, you can just use `Node`.
I would always use `LifecycleNode` action for lifecycle nodes, just for clarity.

See:
https://github.com/ros2/launch_ros/blob/c86ee0b7c74af84c3b54eebede1501b7735431df/launch_ros/launch_ros/actions/lifecycle_node.py#L39-L61