ros2/launch_ros #36: [ros2launch] Optional 'package_name' when full path is provided

artivis, 2019-06-21:
## Feature request

#### Feature description

Ability to call `launch` omitting the `package_name` argument given that the launch file full path is provided.

E.g.
```terminal
$ ros2 launch ./my_launch.launch.py
```

Beside mimicking `roslaunch`, this feature comes handy when one wants to run a launch file that is not in a proper ROS package. Even more so since launch files are now Python scripts.
#### Implementation considerations

I do not know about the difficulties here, there seems to be a somewhat related [todo comment](https://github.com/ros2/launch_ros/blob/master/ros2launch/ros2launch/command/launch.py#L56)  tho.

ivanpauno, 2020-08-05:
@artivis this already works.
I think that the only issue is that the help doesn't show the option correctly.

Can you confirm?

jacobperron, 2020-08-05:
It has been working for me; also with launch files with different front-ends.

clalancette, 2020-08-20:
I think this is working now, and its been a long time since it was opened.  Sorry for the delay in responding.  I'm going to close this out, but if this is still a problem for you please feel free to reopen.

closed 2020-08-20 by clalancette