ros2/launch_ros #38: launch_ros crashes on list-valued parameter

rotu, 2019-07-02:
In trying to use `launch_ros` with composable nodes, I got a hard crash trying to process a parameter file. The syntax that it choked on is not even used by my composable node, but it looks like `to_parameters_list` crashed the entire launch file nonetheless.

Crash trace:
```
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7f391403b668>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7f391403b668>' ✓ '<launch.event_handlers.on_process_start.OnProcessStart object at 0x7f391410a748>'
/opt/ros/master/install/lib/python3.6/site-packages/launch_ros/utilities/to_parameters_list.py:49: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  context, normalize_parameter_dict(yaml.load(f))
Executing <Handle <TaskWakeupMethWrapper object at 0x7f39140b23a8>(<Future finis...events.py:295>) created at /usr/lib/python3.6/asyncio/queues.py:74> took 1.263 seconds
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch/launch_service.py", line 350, in run
    self.__loop_from_run_thread.run_until_complete(run_loop_task)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch/launch_service.py", line 240, in __run_loop
    await process_one_event_task
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 126, in send
    return self.gen.send(value)
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch/launch_service.py", line 177, in _process_one_event
    await self.__process_event(next_event)
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch/launch_service.py", line 197, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch/action.py", line 60, in visit
    return cast(Optional[List[LaunchDescriptionEntity]], self.execute(context))
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch_ros/actions/load_composable_nodes.py", line 171, in execute
    self._load_in_sequence(self.__composable_node_descriptions, context)
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch_ros/actions/load_composable_nodes.py", line 151, in _load_in_sequence
    self._load_node(next_composable_node_description, context)
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch_ros/actions/load_composable_nodes.py", line 114, in _load_node
    context, composable_node_description.parameters
  File "/opt/ros/master/install/lib/python3.6/site-packages/launch_ros/utilities/to_parameters_list.py", line 60, in to_parameters_list
    raise RuntimeError('invalid parameter value {}'.format(repr(value)))
RuntimeError: invalid parameter value (-7, -34, -47)
```

Relevant part of parameter file:
```
bno055_driver:
  ros__parameters:
    port: /dev/imu
    frame_id: imu_link
    frequency: 30.0
    self_manage: true
    use_magnetometer: false

    angular_velocity_stdev: 0.01
    linear_acceleration_stdev: 0.0015
    magnetic_field_stdev: 0.0000005
    orientation_stdev: 0.000001
    # these calibration values need to be determined on a per-device basis.
    calibration/accelerometer_offset: [-7, -34, -47]
    calibration/magnetometer_offset: [275, -193, -104]
    calibration/gyroscope_offset: [-3, -3, -1]
    calibration/accelerometer_radius: 1000
    calibration/magnetometer_radius: 842
```

rotu, 2019-07-02:
This issue is also mentioned here: https://github.com/ros2/rclcpp/issues/715#issuecomment-490425249

wjwwood, 2019-07-25:
Which version of launch are you using? master or Dashing or Crystal maybe?

rotu, 2019-07-25:
I'm using master

flynneva, 2019-08-09:
have you tried adding output='screen' to each of your composablenodes?

eric1221bday, 2019-09-22:
Might need to get OP to re-check, but perhaps this is resolved by https://github.com/ros2/rclpy/pull/389/?

I have been able to get loading lists from YAML file via launch files work with this fix.

closed 2019-09-23 by rotu