ros2/launch_ros #41: Make OnStateTransition work with the new ComposableNode descriptions

ralph-lange, 2019-07-18:
## Feature request

#### Feature description
In Dashing, the ComposableNodeContainer action and the corresponding ComposableNode description were introduced. A ComposableNode description also takes a managed node (aka lifecycle node). However, configuring and activating a lifecycle node by a ComposableNode description (instead of a LifecycleNode action) fails:

```Python
on_inactive_node_B_handler = launch.actions.RegisterEventHandler(
        launch_ros.event_handlers.OnStateTransition(
            target_lifecycle_node=node_B,
            goal_state='inactive',
            entities=[configure_node_A]))

> Caught exception in launch (see debug for traceback): OnStateTransition 
  requires a 'LifecycleNode' action as the target
```

Using a matcher is also not possible but gives an exception. 

```Python
on_inactive_node_B_handler = launch.actions.RegisterEventHandler(
        launch_ros.event_handlers.OnStateTransition(
            matcher=launch_ros.events.lifecycle.matches_node_name("/example/node_B"),
            goal_state='inactive',
            entities=[configure_node_A]))

> ... return lambda action: action.node_name == node_name
  AttributeError: 'Shutdown' object has no attribute 'node_name'
```

When calling matches_action with a ComposableNode description is simply does nothing:

```Python
    configure_node_B = launch.actions.EmitEvent(
        event=launch_ros.events.lifecycle.ChangeState(
            lifecycle_node_matcher=launch.events.matchers.matches_action(node_B),
            transition_id=lifecycle_msgs.msg.Transition.TRANSITION_CONFIGURE,
        )
    )
```

Probably, the whole matching and event system for lifecycle action and events has to be extended for ComposbleNode descriptions.

As a starting for experiments, you may use the damped_pendulum_with_transport_delay_as_composed_node.launch.py of the fmi_adapter_examples package, cf. https://github.com/boschresearch/fmi_adapter_ros2/blob/0.1.5/fmi_adapter_examples/launch/damped_pendulum_with_transport_delay_as_composed_node.launch.py and remove comments in lines 90 to 103 and 106 to 108.

JWhitleyWork, 2020-04-06:
Has there been any movement on this?

vincentrou, 2020-12-10:
This issue seems to be linked to this ongoing refactoring : https://github.com/ros2/design/pull/272#issuecomment-669522706

ipa-cmh, 2022-05-06:
I have an experimental, heavily refactored prototype for the launch_ros composition part in https://github.com/ipa-cmh/launch_ros_experimental that does enable this.
It changes quite a lot of things from an architectural standpoint:
* **ComposableNodeContainer:**
  Listens to LoadNodeEvent, then loads the node via request specified in event.
* **ComposableNode now an Action:**
  It waits for the specified container to become available.
  Then it triggers loading its component by emitting an event to the ComposableNodeContainer that will then load the Node.
* **ComposableLifecyleNode inherits from ComposableNode:**
  It waits for the specified container to become available.
  Then it triggers loading its component by emitting an event to the ComposableNodeContainer that will then load the Node.
  It has the same Event interface as LifecycleNode.

It breaks the existing API, so this probably not mergeable, just wanted to point out it exists.

You can use it like this:
```python

import launch
from launch_ros_experimental.actions import ComposableNode, ComposableNodeContainer, ComposableLifecycleNode
from launch_ros.actions import LifecycleTransition
from lifecycle_msgs.msg import Transition

def generate_launch_description():
    """Generate launch description with multiple components."""
    container = ComposableNodeContainer(
            name='my_container',
            namespace='',
            package='rclcpp_components',
            executable='component_container',
            output='screen',
    )

    node = ComposableNode(
        package='composition',
        component='composition::Talker',
        name='talker',
        target_container='my_container'
    )
    lifecycle_node = ComposableLifecycleNode(
        package='composition',
        component='composition::LifecycleTalker',
        name='lctalker',
        namespace='',
        target_container='my_container'  
    )

    lifecycle_node_1 = ComposableLifecycleNode(
        package='composition',
        component='composition::LifecycleTalker',
        name='lctalker_1',
        namespace='',
        target_container='my_container'  
    )

    lifecycle_transition = LifecycleTransition(
        lifecycle_node_names=['lctalker', 'lctalker_1'],
        transitions_ids=[
            Transition.TRANSITION_CONFIGURE,
            Transition.TRANSITION_ACTIVATE]
    )

    return launch.LaunchDescription([container, node, lifecycle_node, lifecycle_node_1, lifecycle_transition])

```
For testing you can use https://github.com/ipa-cmh/demos/tree/lifecycle_components which has the composition LifecycleTalker component.

ZhenshengLee, 2022-11-08:
https://github.com/ros2/launch/issues/672  is related.

I wonder if I should create all issues in this repo, rather than https://github.com/ros2/launch repo?

ZhenshengLee, 2022-11-18:
> I have an experimental, heavily refactored prototype for the launch_ros composition part in https://github.com/ipa-cmh/launch_ros_experimental that does enable this.

@ipa-cmh thanks for your repo!

I've tryied your repo in my project.

It seems that your repo doesnot support Foxy version am I right?

Which versions do you plan to support?

Thanks.

ZhenshengLee, 2022-11-18:
Also, `launch_ros.events.lifecycle.matches_node_name($LifecycleNodeComponent$)` doesnot work.

leonce-m, 2024-01-29:
Any news on this development? It's been 5 years already and we still can't properly use the launch Event system to manage the lifecycle of composable nodes.

amadeuszsz, 2024-12-16:
Friendly ping 🙏🏻 
Is there any update? I would like to know if this feature might not be feasible for current ROS 2 design or just waiting to be picked up from the backlog.

mjcarroll, 2024-12-16:
There shouldn't be anything that makes this infeasbile, it's just waiting to be picked up.