ros2/launch_ros #43: Namespace not pushed by `PushRosNamespace`

orduno, 2019-07-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 68337f836b9dfbf4bed2b9c3b3ae03d21570cf13
- DDS implementation:
  -  Fast-RTPS

#### Steps to reproduce issue

Given the following `talker_launch.py`

``` python
from launch import LaunchDescription
import launch.actions
import launch_ros.actions

def generate_launch_description():

    return LaunchDescription([
        launch_ros.actions.Node(
            package='demo_nodes_cpp',
            node_executable='talker',
            output='screen',
            node_name='talker')
    ])
```

And the following file with name `include_launch.py` that includes `talker_launch.py`:

```python
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.actions import GroupAction
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import ThisLaunchFileDir
from launch_ros.actions import PushRosNamespace

def generate_launch_description():

    return LaunchDescription([
        GroupAction([
            PushRosNamespace('talker_ns'),
            IncludeLaunchDescription(
                PythonLaunchDescriptionSource([ThisLaunchFileDir(), '/talker_launch.py']))
        ])
    ])

```

#### Expected behavior

We would expect the node name to be: `/talker_ns/talker`.

#### Actual behavior

However, it is: `/talker`

#### Additional information

But, if we provide `node_namespace=''` to the action in `talker_launch.py`, and launch `include_launch.py`, then the node name is `/talker_ns/talker`

wjwwood, 2019-07-25:
@ivanpauno do you have time to have a look at this?

ivanpauno, 2019-07-25:
@orduno Thanks for opening the issue!
https://github.com/ros2/launch_ros/pull/44 should fix the problem.
Could you try it?

orduno, 2019-07-25:
@ivanpauno #44 works for me, thanks!

PR #44 (2019-07-25): Fix PushRosNamespace action
Fixes #43.

Though the namespace is expanded correctly, it isn't added to the command when `node_namespace` argument is `None`:
https://github.com/ros2/launch_ros/blob/68337f836b9dfbf4bed2b9c3b3ae03d21570cf13/launch_ros/launch_ros/actions/node.py#L134-L137

I considered two possible fixes:
- Always add the namespace to the command.
- Change `node_namespace` default to `''`. When doing `node_namespace=None`, `__node:={}` argument won't be passed. If a relative namespace is passed, `ros_namespace` launch config will be used.
For absolute namespaces, `ros_namespace` wi [... cut]
closed 2019-07-25 by ivanpauno