ros2/launch_ros #46: Can't use launch argument for node output

ryanewel, 2019-07-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From apt

#### Steps to reproduce issue
Using a launch file like:
```
import launch
import launch_ros.actions

#Argument names
NODE_NAME = "node_name"
OUTPUT = "output"

def generate_launch_description():
  ld = launch.LaunchDescription([
    launch.actions.DeclareLaunchArgument(
      NODE_NAME,
      default_value="node_name",
    ),
    launch.actions.DeclareLaunchArgument(
      OUTPUT,
      default_value="screen"
    ),
 ])
  node = launch_ros.actions.Node(
      package="package_name",
      node_executable="executable_name",
      output=launch.substitutions.LaunchConfiguration(OUTPUT),
      node_name=launch.substitutions.LaunchConfiguration(NODE_NAME),
  )
  ld.add_action(node)
  return ld
```
Then run:
```
ros2 launch example_launch.py
```
#### Expected behavior
I would expect the output to default to 'screen', but be settable from the cli with `output:=log` or from another launch file that includes this one.
#### Actual behavior
Launch fails with message

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): <launch.substitutions.launch_configuration.LaunchConfiguration object at 0x7f5f04242ac8> is not a valid output configuration
```
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

tobiasneumann, 2020-06-16:
I can reproduce this (with an existing `package_name` and `executable_name`).
* Operating System:
    * Debian 10
* Installation type:
    * source
* Distribution
    * foxy
* Versions:
    * https://github.com/ros2/launch.git
        * dbbd4de641fa9f2116bd71600e2c8fe67f0c24e3
    * https://github.com/ros2/launch_ros.git
        * 54ab32db96e0d813187f24e3c12fb3ea77cf2fa2

hidmic, 2020-06-16:
Indeed. Currently we do not support substitutions as `launch.actions.ExecuteProcess`' output argument (see [constructor](https://github.com/ros2/launch/blob/dbbd4de641fa9f2116bd71600e2c8fe67f0c24e3/launch/launch/actions/execute_process.py#L102)). 

I don't see why we couldn't though, a launch context is available when that output configuration is first used (see [here](https://github.com/ros2/launch/blob/dbbd4de641fa9f2116bd71600e2c8fe67f0c24e3/launch/launch/actions/execute_process.py#L778).

@ryanewel @tobiasneumann would you be willing to contribute a patch?

tobiasneumann, 2020-06-16:
I'm sorry, I can not contribute since my experience and knowledge of python is minimal.

Parv-Maheshwari, 2021-08-18:
@hidmic Are there any plans to include substitutions in the launch description. In my case, I made the namespace , a parameter dependent on the launch argument "robot<id>" which can take vaues like 0,1,2,,,,, . And my namespace is accordingly = "robot"+"robot<id>" i.e. robot0, robot1, etc.

But instead of this behaviour, I am receiving this error - rclpy.exceptions.InvalidNamespaceException: Invalid namespace: namespace must not contain characters other than alphanumerics, '_', or '/':
  '/robot<launch.substitutions.launch_configuration.LaunchConfiguration object at 0x7fc6c9b48a30>'

Please ;et me know in case my use case and example is not clear.