ros2/launch_ros #49: Add more convenient Include action

sloretz, 2019-07-30:
## Feature request

This is a request to make it simpler to include a launch file from a ROS package. Currently it looks like this:

```python3
IncludeLaunchDescription(
    AnyLaunchDescriptionSource(
          [FindPackage('my_package'), '/share/my_package/launch/my_launch.py'])),
```

But it could look like this:

```python3
from launch_ros.actions import IncludeLaunchFile
# ...
IncludeLaunchFile(package='my_package, file='my_launch.py')
```

#### Implementation considerations

This would remove the need to know the full path to the launch file. `FindPackage` might return a path that's shared with many other packages in a merged workspace (like `/opt/ros/dashing`), so it should limit the search to `share/my_package`.

IanTheEngineer, 2019-07-30:
As an alternative to searching for a file directly, it may make sense to detect the root of the package's `install` or `src` directory, and reach into the relative path from there:
```
from launch_ros.actions import IncludeLaunchFile
# ...
IncludeLaunchFile(pkg='my_package', file_path='launch/my_launch.py')
```
similar to[ `rospkg`](http://docs.ros.org/independent/api/rospkg/html/python_api.html):

```
import rospkg

r = rospkg.RosPack()
launch_path = ''.join(r.get_path('my_package'), 'launch/my_launch.launch')
```
or the ROS1 commandline / launchfile syntax with the [rospack comand](http://docs.ros.org/independent/api/rospkg/html/rospack.html)
```
$ `rospack find`/launch/my_launch.launch
```

adityapande-1995, 2022-07-28:
We should move this to ``ros2/launch`` since the ``IncludeLaunchDescription`` action lives there.