ros2/launch_ros #50: Namespace defined on rclcpp::node is not applied

orduno, 2019-07-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - d388bba43fd4681661ab991110e96c01d142cb91
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Given the following modified `MinimalTimer` node from [ROS2 example package](https://github.com/ros2/examples/tree/master/rclcpp/minimal_timer).

```cpp
#include <chrono>
#include <memory>

#include "rclcpp/rclcpp.hpp"

using namespace std::chrono_literals;

class MinimalTimer : public rclcpp::Node
{
public:
  MinimalTimer()
  : Node("minimal_timer", "timer_ns")
  {
    RCLCPP_INFO(this->get_logger(), "The namespace is %s", this->get_namespace());

    timer_ = create_wall_timer(
      500ms, std::bind(&MinimalTimer::timer_callback, this));
  }

private:
  void timer_callback()
  {
    RCLCPP_INFO(this->get_logger(), "Hello, world!");
  }
  rclcpp::TimerBase::SharedPtr timer_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalTimer>());
  rclcpp::shutdown();
  return 0;
}
```

and the following `timer_launch.py`
```python
from launch import LaunchDescription
import launch.actions
import launch.substitutions
import launch_ros.actions

def generate_launch_description():
    """Launch a talker and a listener."""
    return LaunchDescription([
        launch_ros.actions.Node(
            package='examples_rclcpp_minimal_timer',
            node_executable='timer_member_function',
            output='screen')
    ])
```

#### Expected behavior

If launching using `timer_launch.py` we expect the node name to be `/timer_ns/minimal_timer` as a namespace parameter was passed to the node constructor.

#### Actual behavior

However, it is `/minimal_timer`.

#### Additional information

Running the executable does yield the expected node name.

orduno, 2019-07-31:
@ivanpauno would you mind taking a look a this? I think the `PushRosNamespace` feature introduced this bug.

ivanpauno, 2019-08-01:
Thanks for reporting the issue!
https://github.com/ros2/launch_ros/pull/51 should solve the problem

ivanpauno, 2019-08-06:
Solved in https://github.com/ros2/launch_ros/pull/51.

closed 2019-08-06 by ivanpauno