ros2/launch_ros #55: yaml file which passed in launch file doesn't take effect

xhuan28, 2019-08-20:
**Required Info:**

- Operating System:
  - Ubuntu18.04
- Installation type:
  - source
- Version or commit hash:
  - 84cdecd361a3ea274f77f0c1ec2f2c02097fe372
- DDS implementation:
  - fast-rtps
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
Pass the yaml file to launch.py, but it doesn't take effect.
```

#### Expected behavior

#### Actual behavior

#### Additional information

- launch file

```
import os
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    param_file_path = /my/file/path
    container = ComposableNodeContainer(
            node_name='my_container',
            node_namespace='',
            package='rclcpp_components',
            node_executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='my_package',
                    node_plugin='namespace::Class',
                    node_name='my_node',
                    parameters=[param_file_path],),
            ],
            output='screen',
    )

    return launch.LaunchDescription([container])
```

- yaml file:
```
my_node:
  ros__parameters:
    enabled_param1: true
    enabled_param2: true
```

xhuan28, 2019-08-20:
More information:
But the parameters can be set successfully through dictionary one by one. Is there anything wrong with the namespace?
e.g.
```
                ComposableNode(
                    package='my_package',
                    node_plugin='namespace::Class',
                    node_name='my_node',
                    parameters=[param_file_path,                    # no effect
                                         {'enabled_param1': 'true'}],)   # successfully

```

xhuan28, 2019-08-29:
I have found out the root cause of this issue. It is caused by different naming rules between yaml file and direct dictionary parameter. A parameter from a yaml file will combine its name as a string like **"/ns.my_node.ros__parameters.my_param"**, while passing a parameter as a dictionary directly, the name should be "my_param". Parameter system takes these two parameters for different parameters, not the same one. So the parameter from a yaml file will not override the default value to take effect.
I think we need to make the naming rule consistent among different parameter sources. I submit a [PR#58](https://github.com/ros2/launch_ros/pull/58). Please have a review. Thanks.

mabelzhang, 2019-08-29:
Related issue https://github.com/ros2/rclcpp/issues/715

closed 2019-12-17 by ivanpauno