ros2/launch_ros #61: [xml frontend] Can't launch node without name attribute

sloretz, 2019-08-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu bionic
- Installation type:
  - source
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Save the following as `test.launch.xml`

```xml
<?xml version="1.0"?>
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen" args=" ">
  </node>
</launch>
```

Run it with
```
ros2 launch --debug test.launch.xml 
```

#### Expected behavior

Node would successfully launch using whatever name is its default.

#### Actual behavior

An exception is raised.

<details><summary>traceback</summary>

```
$ ros2 launch --debug tf2_ros_py/src/mypkg/launch/test.launch.xml 
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/sloretz/.ros/log/2019-08-29-14-21-16-169948-87cee88a3136-25020
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f768f3ae4e0>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f768f3ae4e0>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f7684a17358>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f768032ddd8>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f768032ddd8>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f7684a17358>'
[DEBUG] [launch]: Traceback (most recent call last):
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_description_sources/frontend_launch_file_utilities.py", line 35, in get_launch_description_from_frontend_launch_file
    return parser.parse_description(root_entity)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/frontend/parser.py", line 102, in parse_description
    actions = [self.parse_action(child) for child in entity.children]
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/frontend/parser.py", line 102, in <listcomp>
    actions = [self.parse_action(child) for child in entity.children]
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/frontend/parser.py", line 85, in parse_action
    return instantiate_action(entity, self)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/frontend/expose.py", line 38, in instantiate_action
    action_type, kwargs = action_parse_methods[entity.type_name](entity, parser)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch_ros/launch_ros/actions/node.py", line 234, in parse
    kwargs['node_name'] = kwargs['name']
KeyError: 'name'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_service.py", line 377, in run_async
    await process_one_event_task
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 126, in send
    return self.gen.send(value)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_service.py", line 287, in _process_one_event
    await self.__process_event(next_event)
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_service.py", line 307, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/action.py", line 112, in visit
    return self.execute(context)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/actions/include_launch_description.py", line 128, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [{}]:

[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [{}]:
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f768032d5c0>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f768032d5c0>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f767c0ecba8>'
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f768032d5c0>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f768f41e4a8>'
Executing <Task finished coro=<LaunchService._process_one_event() done, defined at /home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_service.py:285> result=None created at /home/sloretz/workspaces/ros2-dev0/build/launch/launch/launch_service.py:355> took 0.998 seconds
```
</details>

#### Additional information

The launch file works fine if I add a `name` attribute.

```xml
<?xml version="1.0"?>
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen" name="demo_talker" args=" ">
  </node>
</launch>
```

Similar to #53, but with a different attribute on the `node` tag.

ivanpauno, 2019-09-23:
It should been fixed after https://github.com/ros2/launch_ros/pull/73.
Now `node_name` is the remapping, and `name` refers to the executable name.

hidmic, 2020-02-14:
@ivanpauno @sloretz can this be closed?

ivanpauno, 2020-02-14:
> @ivanpauno @sloretz can this be closed?

Yes, closing!

closed 2020-02-14 by ivanpauno