ros2/launch_ros #64: Node constructor annotations are too opaque

rotu, 2019-09-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - master
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
import launch_ros
help(launch_ros.actions.LifecycleNode)
```

Or use an IDE that shows type annotation information and view the annotation info for the `LifecycleNode` class.

#### Expected behavior
Help will show useful information about the required arguments to the `LifecycleNode` constructor.

#### Actual behavior
Only the information for the `node_name` argument and an opaque `**kwargs` argument is given, as well as an unapologetic caveat.

[image]

```
class LifecycleNode(launch_ros.actions.node.Node)
 |  Action that executes a ROS lifecycle node.
 |  
 |  Method resolution order:
 |      LifecycleNode
 |      launch_ros.actions.node.Node
 |      launch.actions.execute_process.ExecuteProcess
 |      launch.action.Action
 |      launch.launch_description_entity.LaunchDescriptionEntity
 |      builtins.object
 |  
 |  Methods defined here:
 |  
 |  __init__(self, *, node_name:str, **kwargs) -> None
 |      Construct a LifecycleNode action.
 |      
 |      Almost all of the arguments are passed to :class:`Node` and eventually
 |      to :class:`launch.actions.ExecuteProcess`, so see the documentation of
 |      those classes for additional details.
 |      
 |      This action additionally emits some event(s) in certain circumstances:
 |      
 |      - :class:`launch.events.lifecycle.StateTransition`:
 |      
 |          - this event is emitted when a message is published to the
 |            "/<node_name>/transition_event" topic, indicating the lifecycle
 |            node represented by this action changed state
 |      
 |      This action also handles some events related to lifecycle:
 |      
 |      - :class:`launch.events.lifecycle.ChangeState`
 |      
 |        - this event can be targeted to a single lifecycle node, or more than
 |          one, or even all lifecycle nodes, and it requests the targeted nodes
 |          to change state, see its documentation for more details.
 |  
```

#### Additional information

This could be fixed with the either use of the [`merge_args` library](https://pypi.org/project/merge-args/) or the metaprogramming techniques described by its author here: https://chriswarrick.com/blog/2018/09/20/python-hackery-merging-signatures-of-two-python-functions/

deitry, 2019-09-11:
This is what I get as help tooltip in my IDE (VS Code):
```python
class LifecycleNode(node_name: Text, package: SomeSubstitutionsType, node_executable: SomeSubstitutionsType, node_namespace: Optional[SomeSubstitutionsType]=None, parameters: Optional[SomeParameters]=None, remappings: Optional[SomeRemapRules]=None, arguments: Optional[Iterable[SomeSubstitutionsType]]=None, name: Optional[SomeSubstitutionsType]=None, cwd: Optional[SomeSubstitutionsType]=None, env: Optional[Dict[SomeSubstitutionsType, SomeSubstitutionsType]]=None, additional_env: Optional[Dict[SomeSubstitutionsType, SomeSubstitutionsType]]=None, shell: bool=False, sigterm_timeout: SomeSubstitutionsType=LaunchConfiguration(
            'sigterm_timeout', default=5), sigkill_timeout: SomeSubstitutionsType=LaunchConfiguration(
            'sigkill_timeout', default=5), prefix: Optional[SomeSubstitutionsType]=None, output: Text='log', output_format: Text='[{this.name}] {line}', log_cmd: bool=False, on_exit: Optional[Union[
            SomeActionsType,
            Callable[[ProcessExited, LaunchContext], Optional[SomeActionsType]]
        ]]=None, condition: Optional[Condition]=None)
Action that executes a ROS lifecycle node.
```
As ROS2 user, I can't say your way is stricly better since it's easier to look at source file instead of getting through verbosity of your approach. It seems to me that both ways are not good enough to get real help from them.

I guess the real problem is lack of examples that would cover all the features of some technology like launch system - in one place. Or better - centralized feature-full documentation instead of piles of design articles, Q&A, Github issues and so on.

mabelzhang, 2019-10-03:
Thank you for the suggestions! Feel free to open a PR to improve the help message.

On a side note, we are currently actively working on a set of tutorials for ROS 2 to improve the user and developer experience. Keep an eye out :)

rotu, 2019-10-03:
@mabelzhang This seems like low-hanging fruit to improve the user and developer experience. I wrote it up as a github issue so you guys could easily implement it.

dirk-thomas, 2019-10-03:
> This seems like low-hanging fruit to improve the user and developer experience. I wrote it up as a github issue so you guys could easily implement it.

@rotu If it can be so easily implemented please consider to create a pull request for it.

rotu, 2019-10-03:
@dirk-thomas because I thought someone with more familiarity and insight into the codebase would be a better fit for this development.

rotu, 2019-10-03:
https://github.com/ros/rosdistro/pull/22535
I tried to start on this PR and I just don't think I can deal with this dependency hell in using a simple PIP library >.<. I'm just too frustrated with the ROS way of doing Python to fix this issue myself. :-(