ros2/launch_ros #74: Bad interaction between launch frontend and YAML parameters

hidmic, 2019-09-19:
Both launch XML and YAML frontends currently coexist with the YAML parsing of substitution text in `Node` parameters. This has caused us issues in the past (e.g. https://github.com/ros2/launch/pull/226#discussion_r300031818), forcing additional quotes as escape mechanisms. At the time we deemed this as a situation we'd rarely find in the field.

-------

Well,  as you can see in https://github.com/RobotWebTools/rosbridge_suite/pull/433, specifically on commit https://github.com/RobotWebTools/rosbridge_suite/pull/433/commits/d466c89475a068acd3e90a286d633c75d611204c, it's not as rare as we thought. 

There, **I'm forced to surround a `*` with two `'` on each side**.

- **Why a `*` needs this?** Because it's later used as a `Node` parameter and parameters are parsed as YAML, that parses that `*` as a malformed alias (see [PyYAML documentation](https://pyyaml.org/wiki/PyYAMLDocumentation) about aliases). 

- **Why two `'`?** The inner pair prevents `*` from being parsed as a YAML alias, and the outer pair is removed by the frontend for consistency -- otherwise one has to remember that some values given in e.g. XML are parsed using YAML under the hood and some are not.  

IMHO the deeper, unresolved issue that https://github.com/ros2/launch_ros/pull/31 partially addressed is that Substitutions in `launch` are always text and it is up to the receiving end to do any additional parsing, opening the door to things like this.

dirk-thomas, 2019-09-19:
Since this affects how you have to write a launch file I would think this needs to be fixes asap before more packages start using this.

ivanpauno, 2019-09-19:
A think we should have a custom class for a parameter description, instead of using a native type to represent them. There are a bunch of reasons:
- Conditional inclusion of parameters isn't now possible.
  As I commented [here](https://github.com/ros2/launch/issues/290#issuecomment-520823768) it would be much easier if we have a custom class for that.
- If we would have a class that could be constructed like:
  `Parameter('name', VALUE_HERE, type=int)`
  We could avoid all this problem.

Also, we currently don't provide a way of coercing the result of a substitution to another thing that's not an string. So each class is doing a custom thing for doing this conversions (e.g.: for converting from `str` to `bool`). I did a comment about "typed substitutions" in https://github.com/ros2/ros2_documentation/pull/302#discussion_r306325733, though that's not the only solution.

ivanpauno, 2020-08-05:
After https://github.com/ros2/launch_ros/pull/137, the rules used for coercing type after substitutions and by launch frontend are the same.

closed 2020-08-05 by ivanpauno