ros2/launch_ros #75: secondary launch adding components into running containers

dirk-thomas, 2019-09-19:
For modularity the user might want to split a launch file into separate files which are started independently. One example would be a `base_launch.py` file which gets started once and a `dev_launch.py` file which contains stuff which the developer iterates on which is being started / stopped repeatedly.

For nodes running in separate processes these launch files can be started independently. What is not supported at the moment is for both launch files to contribute components into the same process. E.g. the second launch file want to inject a component into a process which was started by the first launch file.

The goal would be to support this use case within launch.

eric1221bday, 2019-09-21:
I've been dealing with this issue recently, and looking at the current implementation of `LoadComposableNodes`. It seems like some simple modifications should suffice to make this use case work. The class takes in a full `ComposableNodeContainer` but only really needs the `node_name` of that container.

Perhaps we can simply modify or add a new argument `target_container_name`, which, if filled, would be the one used to call the service call. That name can be a `str` or maybe even a `Substitution`

eric1221bday, 2019-10-25:
Oh right, so #77 which I authored should be able to deal with this problem

ivanpauno, 2019-12-04:
Fixed by https://github.com/ros2/launch_ros/pull/77.

closed 2019-12-04 by ivanpauno