ros2/launch_ros #88: Invalid keyword argument

dabonnie, 2019-10-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
  - Jenkins build farm - multiple systems (linux, osx, etc)
- Installation type:
  - source
- Version or commit hash:
  - [master repos file](https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos)

[Build farm test link](https://ci.ros2.org/view/nightly/job/nightly_linux-armhf_release/lastCompletedBuild/testReport/test_launch_ros.test.test_launch_ros.frontend/test_node_frontend/test_node_frontend__nlaunch__n______let__n________name___a_string__n________value_____2__5__8____n______let__n________name___a_list__n________value____2__5__8___n______node__n________pkg__demo_nodes_py_n________exec__talker_qos_n________output__screen_n________name__my_node_n________namespace__my_ns_n________args_____number_of_cycles_1__n________param__n________________name__param1_n________________value__ads_n________________name__param_group1_n________________param__n____________________name__param_group2_n____________________param__n________________________name__param2_n________________________value__2_n____________________name__param3_n____________________value___2__5__8__n____________________name__param4_n____________________value__$_var_a_list__n____________________name__param5_n____________________value__$_var_a_string__n____________________name__param6_n____________________value___2___5___8___n____________________name__param7_n____________________value____2____5____8___n____________________name__param8_n____________________value_____2______5______8____n____________________name__param9_n____________________value_________2______________5______________8________n____________________name__param10_n____________________value_____asd______bsd______csd____n____________________name__param11_n____________________value______asd______bsd______csd___n____________________name__param12_n____________________value_____n________________from___home_jenkins_agent_workspace_nightly_linux_armhf_release_ws_src_ros2_launch_ros_test_launch_ros_test_test_launch_ros_frontend_params_yaml_n________env__n________________name__var_n________________value___1__n_/)

#### Steps to reproduce issue

Check out a clean workspace

```
colcon build --packages-up-to test_launch_ros
colcon test --packages-up-to test_launch_ros
```

#### Expected behavior

test/test_launch_ros/frontend/test_node_frontend.py passes

#### Actual behavior

```
TypeError: parse_substitution() got an unexpected keyword argument 'optional'
```

#### Additional information

dabonnie, 2019-10-15:
The fix is fairly straightforward :-)

```diff
diff --git a/launch_ros/launch_ros/actions/node.py b/launch_ros/launch_ros/actions/node.py
index 7b36cb4..bf7b77e 100644
--- a/launch_ros/launch_ros/actions/node.py
+++ b/launch_ros/launch_ros/actions/node.py
@@ -239,7 +239,7 @@ class Node(ExecuteProcess):
         node_name = entity.get_attr('node-name', optional=True)
         if node_name is not None:
             kwargs['node_name'] = node_name
-        package = parser.parse_substitution(entity.get_attr('pkg'), optional=True)
+        package = parser.parse_substitution(entity.get_attr('pkg'))
         if package is not None:
             kwargs['package'] = package
         kwargs['node_executable'] = parser.parse_substitution(entity.get_attr('exec'))
```

dirk-thomas, 2019-10-15:
> The fix is fairly straightforward :-)

@dabonnie How about creating a pull request with that patch then :wink:

dabonnie, 2019-10-15:
> > The fix is fairly straightforward :-)
> 
> @dabonnie How about creating a pull request with that patch then

In progress!

dabonnie, 2019-10-15:
Proposed changes in this PR: https://github.com/ros2/launch_ros/pull/89

wjwwood, 2019-10-15:
I think this was already proposed to be fixed in #83.

dabonnie, 2019-10-15:
> I think this was already proposed to be fixed in #83.

Yeah, thanks for pointing that out!

closed 2019-10-15 by dabonnie