ros2/launch_ros #91: Release 'launch_testing_ros' to Crystal

TSC21, 2019-10-27:
#### Feature description
There's a great potential on the testing framework that would be great to have in Crystal as well, specially for testing packages released for multiple distro versions (in this case, considering Crystal as well)

#### Implementation considerations
I guess it would be as simple as doing a bloom release for Crystal, unless the API doesn't allow it or requires backporting other changes.

@wjwwood @sloretz thanks in advance!

TSC21, 2019-10-27:
I am not sure if [launch_testing](https://github.com/ros2/launch/tree/master/launch_testing) would be the package to use for testing node outputs (which seems to be a dependency from `launch_testing_ros`), though even after installing `ros-crystal-launch-testing` on the system, I still get `launch_test: command not found`.

wjwwood, 2019-11-15:
It's not going to be possible to do this in Crystal because of the features required from the `ros2/launch` repository, which we cannot upgrade without breaking API and disrupting Crystal users.

So I'll close this as won't fix. Sorry.

closed 2019-11-15 by wjwwood