ros2/launch_ros #97: talker_listener_launch_test example is broken

cottsay, 2019-11-25:
Trying to run this launch test:
```
python3 -m launch_testing.launch_test src/ros2/launch_ros/launch_testing_ros/test/examples/talker_listener_launch_test.py
```

Yields:
```
Traceback (most recent call last):
  File "src/ros2/launch_ros/launch_testing_ros/test/examples/talker_listener_launch_test.py", line 72, in test_talker_transmits
    node = launch_context.locals.launch_ros_node
  File "build/launch/launch/launch_context.py", line 137, in __getattr__
    ', '.join(_dict.keys())
AttributeError: context.locals does not contain attribute 'launch_ros_node', it contains: []
```

teutobod, 2019-12-12:
Any updates here? I'm stuck to the same issue.

jacobperron, 2019-12-16:
I can reproduce the issue.

Interestingly, if you run the test with the CLI tool `ros2 test`, then there is no error:

    ros2 test src/ros2/launch_ros/launch_testing_ros/test/examples/talker_listener_launch_test.py

jacobperron, 2019-12-17:
I guess it kind of makes sense, since the ROS node is specific to `launch_ros` and not `launch`; the node is started by the default `launch_ros` description here:

https://github.com/ros2/launch_ros/blob/c37340d04803df34d6cc8d83436cc448fed62d6a/launch_ros/launch_ros/default_launch_description.py#L80-L92

I think either the README should be updated to use `ros2 test`, or the example launch file should be updated to use the `get_default_launch_description`.

jacobperron, 2019-12-17:
Updating the launch file fixes the issue for launching with `launch_test`, e.g.:

```python
    ld = launch_ros.get_default_launch_description()
    ld.add_action(talker_node)
    ld.add_action(listener_node)
    # Start tests right away - no need to wait for anything
    ld.add_action(launch_testing.actions.ReadyToTest())
```

But, then two nodes are created if we run with `ros2 test` and we get the following warning:

```
[WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

Seems like a different approach is required.

hidmic, 2020-01-02:
There's no good solution here really, besides not mixing up `launch_test` and `ros2 test`. The former just runs the test, the latter injects `launch_ros` default launch description. There was a discussion about this in https://github.com/ros2/launch/issues/208. I still think tools should not inject anything, but consensus back then was to reduce boilerplate as much as possible.

jacobperron, 2020-01-03:
See #112 for a fix of the example.

Longer term, I think we need to remove the injection from the CLI tool, or be able to detect that the preamble has already been included to avoid including it more than once.

stonier, 2020-01-23:
Got burned by this last night as well.

+1 for removing the injection. Reasoning:

* Prefer to have one launch test file format, not two (e.g. the current launch_test file is broken with launch_test, which makes it something else).
* Prefer not to *depend* on client tooling (i.e. `ros2 test`). That should be there to provide conveniences, not dependencies.

jacobperron, 2020-01-30:
I've fixed the example in https://github.com/ros2/launch_ros/pull/121 and opened a ticket to discuss removing the injected launch description: #120

PR #121 (2020-01-31): Fix launch_testing_ros example
Fixes #97.

The example should not be using the underlying launch_ros ROS context or ROS node since it is an
implementation detail. Instead, the example initailizes it's own context and ROS node.
commit 93b242cc (2020-01-31): Fix launch_testing_ros example (#121)

The example should not be using the underlying launch_ros ROS context or ROS node since it is an
implementation detail. Instead, the example initailizes it's own context and ROS node.

Fixes #97.
closed 2020-01-31 by jacobperron