ros2/launch_ros #105: Passing argument to launch file yields wrong RCL warning

Karsten1987, 2019-12-11:
## Bug report

I am receiving a misleading/wrong warning when trying to pass arguments to the `ros2 launch` command.

**Required Info:**

- Operating System:
  - OSX
- Installation type:
  - Eloquent source
- Client library (if applicable):
  - rcl ?

#### Steps to reproduce issue
Given a launch file with any `LaunchArgument`
```
    launch_description = LaunchDescription([
        DeclareLaunchArgument(
            'my_argument',
            default_value = 'some_default',
            description = 'some description'),
```
Then launching the launch file as follows:
```
ros2 launch my_launchfile.launch.py my_argument:=some_custom_value
```
#### Expected behavior

Launch file started without any complains and launch argument successfully remapped.

#### Actual behavior
```
[WARN] [rcl]: Found remap rule 'my_argument:=some_custom_value'. This syntax is deprecated. Use '--ros-args --remap my_argument:=some_custom_value' instead.
```

Similarly, when using the proposed syntax, the substitution does not take place.

Please feel free to transfer to ticket to a more appropriate repo if applicable.

jacobperron, 2019-12-13:
I also ran into the same issue.

jacobperron, 2019-12-16:
I think this was fixed by https://github.com/ros2/launch_ros/pull/106. @ivanpauno Can you close this ticket if I'm correct?

closed 2019-12-17 by ivanpauno