ros2/launch_ros #109: arguments = [--ros-args --log-level WARN] has not expected behavior

mganglb, 2019-12-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  -  binaries 

#### Steps to reproduce issue
- making a launch file (Python) in a pkg
- building
- ros2 launch <pkg> <launch-file>

`

    xyz_cmd= Node(
        package=package;
        node_executable='node_executable',
        node_name=node_name'
        node_namespace = node_namespace ,
        output='screen',
        parameters = [];
        arguments = [('__log_level:=WARN')])
`
**arguments = [('__log_level:=WARN')]** 

leads to: " [WARN] [rcl]: Found log level rule '__log_level:=WARN'. This syntax is deprecated. Use '--ros-args --log-level WARN' instead.
[joint_state_publisher-1] [WARN] [rcl]: Found log level rule '__log_level:=WARN'. This syntax is deprecated. Use '--ros-args --log-level WARN' instead."

`
**arguments = [(--ros-args --log-level WARN)]** 
`
This does not work
#### Additional information
In the launch.py file the log-level has to be defined like that
`
arguments = ['--ros-args', '--log-level', 'DEBUG'],
`

hidmic, 2020-01-02:
> `**arguments = [(--ros-args --log-level WARN)]**`
This does not work

I have the slight impression this isn't valid Python syntax.

> In the launch.py file the log-level has to be defined like that
`arguments = ['--ros-args', '--log-level', 'DEBUG'],`

For the sake of simplicity, the syntax suggested in warnings assumes you're running in a shell environment, which takes care of tokenizing your arguments. Depending on how you run your processes, you may have to provide arguments in a different way. For your particular use case, and in general when programmatically executing a process from a process, arguments must be provided one by one as shown (because ultimately the OS demands it, e.g. check [`execve()`](http://man7.org/linux/man-pages/man2/execve.2.html) in Linux). 

I'll close this one out for now. Feel free to continue the discussion afterwards.

closed 2020-01-02 by hidmic

mganglb, 2020-01-03:
Thanks for your response!

> **arguments = [(--ros-args --log-level WARN)]**
This does not work

This is ofc not correct, I can’t even reproduce my thoughts/error anyway atm. 

Still, I am a bit confused with the logging system. If I use a launch.py file to start nodes, with debug level warn or info”, “info” msgs are not displayed on the console. However, they appear after or while the node get closed again. I found this behavior with python and c++ nodes.
If I run in a shell the logging is working as expected.
``
    my_logger = node.get_logger()

    my_logger.info('INFO')                  	# visible on console when node shuts down

    my_logger.warn("WARN")                  	# visible on console

    my_logger.debug("DEBUG")                	# visible on console

``

In addition: If I set the logging level to debug, the INFO msgs appears on the console after the WARN msgs.

Anyway, maybe this is some default behavior; I am not experienced enough in ROS2 nor the logging system. Just wanted to mention it.

dirk-thomas, 2020-01-03:
@mganglb what you are seeing is that debug/info goes to `stdout` which is by default a fully buffered stream while warn/error goes to `stderr` which is line buffered. See ros2/rcutils#196 for a proposal to change all logging to a single stream.