ros2/launch_ros #114: LaunchConfiguration is not read from ComposableNode when included inside a group action

jacobperron, 2020-01-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - c5e2aa2bf0eb60d61c66dbe0a6da365230b3387d
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Create the following two launch files:

`foo.launch.xml`
```
<launch>
  <group>
    <include file='bar.launch.py'>
    </include>
  </group>
</launch>
```

`bar.launch.py`
```
import launch
import launch_ros

def generate_launch_description():
    return launch.LaunchDescription([
        launch.actions.DeclareLaunchArgument(name='bar_arg', default_value='True'),
        launch_ros.actions.ComposableNodeContainer(
            package='rclcpp_components', node_executable='component_container',
            node_name='my_container', node_namespace='',
            composable_node_descriptions=[
                launch_ros.descriptions.ComposableNode(
                    package='demo_nodes_cpp',
                    node_plugin='demo_nodes_cpp::Talker',
                    parameters=[{'use_sim_time': launch.substitutions.LaunchConfiguration('bar_arg')}]
                ),
            ]
        ),
        # launch_ros.actions.Node(
        #     name='talker', package='demo_nodes_cpp', node_executable='talker', output='screen',
        #     parameters=[{'use_sim_time': launch.substitutions.LaunchConfiguration('bar_arg')}]),
    ])
```

Launch `foo.launch.xml`:

    ros2 launch foo.launch.xml

#### Expected behavior

No launch errors.

#### Actual behavior

Launch error:

```
[INFO] [launch]: All log files can be found below /home/jacob/.ros/log/2020-01-16-13-53-42-380127-warner-9368
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [9379]
[... 58 lines cut ...]
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fb51d289c88>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fb51d289c88>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7fb51d2965f8>'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.SignalProcess'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fb51d289c88>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7fb5204265c0>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fb51d289c88>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7fb52dde0e10>'
Executing <Task finished coro=<LaunchService._process_one_event() done, defined at /opt/ros/eloquent/lib/python3.6/site-packages/launch/launch_service.py:289> result=None created at /opt/ros/eloquent/lib/python3.6/site-packages/launch/launch_service.py:359> took 0.743 seconds
[DEBUG] [launch]: processing event: '<launch.events.process.signal_process.SignalProcess object at 0x7fb51d343748>'
[DEBUG] [launch]: processing event: '<launch.events.process.signal_process.SignalProcess object at 0x7fb51d343748>' ✓ '<launch.event_handler.EventHandler object at 0x7fb51d296908>'
[INFO] [component_container-1]: sending signal 'SIGINT' to process[component_container-1]
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0x7fb52041cf28>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0x7fb52041cf28>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7fb51d2965c0>'
[INFO] [component_container-1]: process has finished cleanly [pid 9462]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessExited'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7fb52041cdd8>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7fb52041cdd8>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x7fb51d296518>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7fb52041cdd8>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x7fb51d296630>'
</pre>
</details>

#### Additional information

If we remove the `<group>` tag in `foo.launch.xml`, then there is no issue.

If we use a Node action instead of a composable node, then there is no issue.

wjwwood, 2020-01-17:
Maybe you need to use the `scoped` argument for the `GroupAction`? I need to look in more detail at why you are using the group, but I guess the scope option needs to be flipped or the scope mechanism has a bug in it or something.

wjwwood, 2020-01-17:
Maybe it has to do with the delayed evaluation of the substitution? I'm not sure...

wjwwood, 2020-01-17:
I think since the load of the composable nodes is done asynchronously (with an event handler) the group has already ended and undone the scope:

https://github.com/ros2/launch_ros/blob/c5e2aa2bf0eb60d61c66dbe0a6da365230b3387d/launch_ros/launch_ros/actions/composable_node_container.py#L73

I'm not sure how to fix this, a few options:

- perform the substitutions in `ComposableNodeContainer::execute()` before registering the event
- change how the containers are loaded to avoid the delay in evaluating the substitutions (i.e. do not use an event)
- store the launch config state and make sure it is used for the evaluation in the event handler (sort of like lambda capture or something)

jrgnicho, 2020-05-06:
Also experiencing the same issue with ComposableNodeContainer where my python launch file looks like this
```
def generate_launch_description():
   
    container =  ComposableNodeContainer(
            node_name= 'perception',
            node_namespace= launch.substitutions.LaunchConfiguration('global_ns'),
            package= 'rclcpp_components',
            node_executable = 'component_container',
            composable_node_descriptions = [
                ComposableNode(
                package='crs_perception',
                node_plugin='crs_perception::LocalizeToPart',
                node_name='part_localization',
                node_namespace = launch.substitutions.LaunchConfiguration('global_ns')),
                ],
            output = 'screen'
        )    
    
    return launch.LaunchDescription([
        launch.actions.DeclareLaunchArgument('global_ns', default_value = ['/my_ns']),
        container])
```
It runs fine when I call it from the command line
but when I call it from an xml launch file 
```
	    <include file="$(find-pkg-share crs_application)/launch/perception.launch.py">
	    	<arg name="global_ns" value="$(var group_ns)"/>
	    </include>
```
It errors out saying that the `launch configuration 'global_ns' does not exist`.  Removing the argument has no effect either

jacobperron, 2020-08-11:
I did some investigation and I can confirm the launch configuration dropped by the event handler:

https://github.com/ros2/launch_ros/blob/8c05ebb4be52dcd360e47694875643e4f53a010e/launch_ros/launch_ros/actions/composable_node_container.py#L95-L105

Looking through the commit history, the whole event handler strategy for loading seems vestigial; it was originally used in the LoadComposableNodes action, but since moved to the container action (see https://github.com/ros2/launch_ros/pull/16). I don't think it is necessary to use an event and we can simply return the load node action. See https://github.com/ros2/launch_ros/pull/170 for proposed fix.

commit c08b666f (2020-08-10): Do not use event handler for loading composable nodes

Fixes #114.

Due to the asynchronous nature of the LoadComposableNodes action, an event handler causes the launch configuration is popped if ComposableNodeContainer appears inside a group or include action.
It seems to me we can simply return the load node action, which will get executed after the ComposableNodeContainer action. The use of an event handler is vestigial of a refactoring done in https://github.com/ros2/launch_ros/pull/16, and doesn't appear to be necessary.
commit 6f9ea340 (2020-08-10): Do not use event handler for loading composable nodes

Fixes #114.

Due to the asynchronous nature of the LoadComposableNodes action, an event handler causes the launch configuration is popped if ComposableNodeContainer appears inside a group or include action.
It seems to me we can simply return the load node action, which will get executed after the ComposableNodeContainer action. The use of an event handler is vestigial of a refactoring done in https://github.com/ros2/launch_ros/pull/16, and doesn't appear to be necessary.
commit dcb2637d (2020-08-11): Do not use event handler for loading composable nodes

Fixes #114.

Due to the asynchronous nature of the LoadComposableNodes action, an event handler causes the launch configuration is popped if ComposableNodeContainer appears inside a group or include action.
It seems to me we can simply return the load node action, which will get executed after the ComposableNodeContainer action. The use of an event handler is vestigial of a refactoring done in https://github.com/ros2/launch_ros/pull/16, and doesn't appear to be necessary.
PR #170 (2020-08-17): Do not use event handler for loading composable nodes
Fixes #114.
Depends on #166 
Depends on https://github.com/ros2/launch/pull/449

Due to the asynchronous nature of the LoadComposableNodes action, an event handler causes the launch configuration is popped if ComposableNodeContainer appears inside a group or include action.
It seems to me we can simply return the load node action, which will get executed after the ComposableNodeContainer action. The use of an event handler is vestigial of a refactoring done in https://github.com/ros2/launch_ros/pull/16, and doesn't appear to be necessary.
closed 2020-08-17 by jacobperron