ros2/launch_ros #119: Error when set default_value to []

mganglb, 2020-01-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - eloquent

#### Steps to reproduce issue
```
    list_foo= LaunchConfiguration('list_foo')

    declare_namespace_cmd = DeclareLaunchArgument(
        'list_foo',
        default_value=[],
        description='a list')
```

if using an empty list as default_value an error occurs

#### Expected behavior
no error. since in python this is possible:
`node.declare_parameter('list_foo', [], ParameterType.PARAMETER_STRING_ARRAY)`

#### Actual behavior
error (i can post tomorrow the actual error)

regards magnglb

jacobperron, 2020-01-31:
I'm not having any issue setting a default value of `[]` (which evaluates to the empty string). Here's my test:

```python
import launch

def generate_launch_description():
    return launch.LaunchDescription([
        launch.actions.DeclareLaunchArgument(name='foo_arg', default_value=[]),
        launch.actions.ExecuteProcess(
            cmd=['echo', launch.substitutions.LaunchConfiguration('foo_arg')],
            output='screen'
        )   
    ]) 
```

Can you please provide a complete launch script that reproduces the issue?

jacobperron, 2020-03-05:
@mganglb Friendly ping. If you are still experiencing this error, please comment with more information and we can re-open this issue.

closed 2020-03-05 by jacobperron