ros2/launch_ros #124: Output a warning when launching multiple nodes with the same name

emersonknapp, 2020-02-18:
## Feature request
Node name uniqueness is not enforced in ROS2. It seems to be uncertain whether-or-when it will be so. In the meantime, there is a lot of user confusion that can happen around having multiple nodes with the same name, since much of ROS2 assumes they are uniquely named.

One user-helping mitigation we can introduce is: Within a single invocation of `ros2 launch`, log a warning message for duplicate node names before actually starting them. This will highlight one source of duplicate node names.

Limitations:
* this does nothing to help in separately-launched launchfiles (e.g. other robots on the network or separate shells)
* this probably can't introspect the default names if users do not specify a node name on the Node action

#### Implementation considerations
I'm not yet familiar enough with this code to understand how to implement. I assume at some point launch has a fully substituted flattened list of things to start, at which point we'd have all the node names expanded out and can detect duplicates.

jacobperron, 2020-02-26:
It's not necessarily even decided that node names should be unique. I think we should resolve https://github.com/ros2/design/issues/187 so that it's clear what kinds of assumptions can be made about node name uniqueness. Based on the discussion so far, I think the consensus is that there should be some way to uniquely identify nodes, but whether this is based on their name is TBD.

emersonknapp, 2020-02-26:
Right now - non-unique node names cause actual problems in usage. Our intent is a low-impact and minimally invasive addition of warnings that can help users understand/resolve those problems in their system.

The design discussion seems around node name uniqueness seems far from complete, and I agree that we should agree on a well-reasoned approach. However, my understanding is that's not going to happen for Foxy, and then users will be at least a year away from any resolution of confusion caused by overlapping node names. The hope with this issue and https://github.com/ros2/ros2cli/issues/453 is to mitigate the impact of the existing problems by making the user more informed without changing the functionality.

closed 2020-03-10 by ivanpauno