ros2/launch_ros #129: Provide Documentation on how to use XML API

saibotUrczf, 2020-03-11:
## Feature request

#### Feature description

Provide an example on how to use the XML API, i.e., if I have a launch file, how can I get it running? Do I need to create a package, do I need to add it in `package.xml` whatsoever...

Especially since the release of Eloquent Elusor, the launch system became way more practical (imho) and is one of the great features of ROS1. Nonetheless, I struggle to get it running. I already asked on [answers.ros](https://answers.ros.org/question/341128/ros2-launch-xml-example/), receiving no answer.

#### Implementation considerations

ivanpauno, 2020-03-20:
I replied in the ros answers question.

The best source of documentation about XML launchfiles is the migration guide: https://index.ros.org/doc/ros2/Tutorials/Launch-files-migration-guide/.

I'm closing this issue. If you have any specific documentation improvement in mind, we can reopen the issue and discuss it.

closed 2020-03-20 by ivanpauno

sjauhri, 2020-09-23:
Just had the same experience trying to migrate ROS1 XML launch files to ROS2 eloquent.
Was only getting the error "[ERROR] [launch]: Caught exception in launch (see debug for traceback): The launch file may have a syntax error, or its format is unknown"

Found out later that XML launch requires additional packages: 
sudo apt install ros-eloquent-launch*

Would be really really useful if this could be added to the migration guide: https://index.ros.org/doc/ros2/Tutorials/Launch-files-migration-guide/

clalancette, 2020-09-23:
> Would be really really useful if this could be added to the migration guide: https://index.ros.org/doc/ros2/Tutorials/Launch-files-migration-guide/

Please feel free to open a pull request against https://github.com/ros2/ros2_documentation ; we're happy to receive and review contributions.  Thanks.