ros2/launch_ros #136: Command substitution: special characters inside XML comments causes launch to fail

jacobperron, 2020-03-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - from source 
- Version or commit hash:
  - 4333910ba41977f3ab9867082c86a8698ada3a16 

Specifically, I've noticed that colons (`:`) and tab characters are not working.

#### Steps to reproduce issue

I ran into this issue trying to pass a URDF from xacro to robot_state_publisher, like the [example here](https://github.com/ros/robot_state_publisher/blob/f877cbf4399d8c497b946376664bd76b0c192920/launch/rsp-launch-xacro-command-subst.xml). For some reason, if the URDF/xacro file contains a colon (`:`) or a tab character inside of an XML comment, then launch fails.

Here's my example URDF file `foo.urdf`:

```xml
<?xml version="1.0" ?>
<robot>
  <!--: -->
</robot>
```

And my launch file:

```xml
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen">
    <param name="foo" value="$(command 'cat foo.urdf')" />
  </node>
</launch>
```

#### Expected behavior

The launch file succeeds and we see the talker node chatting.

#### Actual behavior

Error:

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): mapping values are not allowed here
  in "<unicode string>", line 3, column 7:
      <!--: -->
          ^
```

#### Additional information

The error seems to specifically occur if there is a colon followed by a space. E.g. making the following change works fine:

```diff
-      <!--: -->
+      <!--:-->
```

Replacing the colcon with a tab character also triggers the same error:

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): while scanning for the next token
found character '\t' that cannot start any token
  in "<unicode string>", line 3, column 7:
      <!--	-->
          ^
```

ivanpauno, 2020-08-05:
This can be fixed now by avoiding using `yaml` conversions:

```xml
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen">
    <param name="foo" value="$(command 'cat foo.urdf')" type="str"/>
  </node>
</launch>
```

The fix was introduced in https://github.com/ros2/launch_ros/pull/137.

closed 2020-08-05 by ivanpauno

karl-schulz, 2022-01-28:
Is there also a fix available for python launch files? Having special characters like `:` or `#` still breaks the parameter parsing in that case...

Example
```
Node(
            package='robot_state_publisher',
            executable='robot_state_publisher',
            name='robot_state_publisher',
            output='screen',
            parameters=[{
                'use_sim_time': use_sim_time,
                'robot_description':Command(['xacro',' ', xacro_path])
            }]),
```
... results in the same error as in the examples in this issue.

See also my comment in https://github.com/ros/robot_state_publisher/issues/155#issuecomment-1023372556