ros2/launch_ros #142: Add `push_ros_namespace` alias to `push-ros-namespace`

clalancette, 2020-04-16:
## Bug report

`push-ros-namespace` is the current decorator used for pushing a ROS namespace within a Launch GroupAction.  However, we think it is more aesthetically nice to use `push_ros_namespace`, so we should add that decorator as well.  See https://github.com/ros2/ros2_documentation/pull/539/files#r388486312 for more details.

closed 2021-06-29 by hidmic

HappySamuel, 2022-03-17:
Hi

I think there's a need to edit this guidance, else everyone will still using `push_ros_namespace` and stumbled until they found this `push-ros-namespace` solution
[image]

Best,
Samuel

clalancette, 2022-03-17:
> I think there's a need to edit this guidance, else everyone will still using `push_ros_namespace` and stumbled until they found this `push-ros-namespace` solution

Sorry, it's not clear to me what the issue is.  We prefer `push_ros_namespace` in Rolling, and that is what is shown in the example you are talking about.  For Foxy and Galactic, we only have `push-ros-namespace`, and the tutorials reflect that.

Either way, if you have a question about the usage please open a question on https://answers.ros.org.  If you think you've genuinely  found a bug, please either open a new issue on https://github.com/ros2/ros2_documentation (to update the documentation) or https://github.com/ros2/launch_ros.

HappySamuel, 2022-03-18:
Thanks for the clarification, i didn't notice that is for Rolling. Sorry for causing trouble.

PR #250 (2021-06-29): Add `push_ros_namespace` alias to `push-ros-namespace`
Closes #142

Once this is approved, I will update the documentation to use the alias with the underscores (https://github.com/ros2/ros2_documentation/pull/539/files#r409568794).