ros2/launch_ros #149: No tab autocomplete or suggesting for xml launch files

Michael-Equi, 2020-05-13:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  -  binaries 
- Version or commit hash:
  - 0.9.6-1bionic.20200121.220323 for launch and 0.9.6-1bionic.20200121.220435 for launch xml
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Install binaries from apt and create an xml launch file in a custom package. Build and source package then perform ros2 launch package_name tab-tab. 

#### Expected behavior
The autocomplete list should show any xml launch files available in that package.

#### Actual behavior
Only .py launch files are shown. If the name of the xml launch file is input into the ros2 launch command and run it launches fine but the xml does not show up when trying to perform tab-completion.

Not sure if this is a bug in existing code or is just not implemented yet but it sure is annoying when you have a lot of xml launch files in a package. I have not tested this issue with yaml launch files.

hidmic, 2020-05-21:
@Michael-Equi IIRC autocomplete for non Python launch files is not supported yet. Would you be willing to contribute said feature?

Michael-Equi, 2020-05-21:
Not exactly sure where I should start, if you could point me in a direction I can see if I have the time to contribute it.

danwblackburn, 2020-09-18:
I was having the same issue in foxy, but I think that I figured out what's wrong. It comes from the naming scheme of the XML launch files. Auto tab completion works fine for me as long as the XML launch files end with ``launch.xml``. Which is consistent with what should be happening now that I read the code [here](https://github.com/ros2/launch_ros/blob/2e64ea5c0d6472543417444513827f696210d877/ros2launch/ros2launch/api/api.py#L79-L81). However in ROS1, XML files normally had just the ``.launch`` extension so that it was compatible with roslaunch. However, in ROS2 a file ending in just ``.launch`` isn't being recognized as a valid launch file. @Michael-Equi If you're still tracking could you verify that you were ending the files in ``.launch`` instead of ``launch.xml``

I don't know if this is on purpose or not because I haven't found any documentation discussing ROS2 xml launch file names, but I can't think of any scenario where someone would want to end a file with ``.launch`` but not have it be a valid launch file.

The solution should be relatively simple by just adding ``.launch`` to ``is_launch_file.extensions``. I'll submit a pull request once I can verify that this is the only thing causing the issue.

hidmic, 2020-10-21:
Circling back. @danwblackburn your analysis is correct, but I think the problem stems from how those extensions are being discovered. [`launch.frontend.Parser.get_available_extensions()`](https://github.com/ros2/launch/blob/d648e19590d425e5d38411fe83fdb3b59f56d586/launch/launch/frontend/parser.py#L111-L115) relies on frontend entrypoint names (e.g. `xml`, `yaml`). Perhaps each parser could provide a list of well-known file extensions.

CC @ivanpauno for feedback.

ivanpauno, 2020-10-21:
Adding `.launch` as an extension that auto-completion recognizes sounds like a reasonable thing to do.

closed 2021-08-25 by hidmic