ros2/launch_ros #151: ComposableNodeContainer launches all ComposableNodes on the same thread

KenYN, 2020-06-11:
## Bug report

**Required Info:**
- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Eloquent
- DDS implementation:
  - NA
- Client library (if applicable):
  - NA

#### Steps to reproduce issue
Taking [`demos/composition`](https://github.com/ros2/demos/tree/eloquent/composition) as a base, change the `launch/composition_demo.launch.py` container line to:
```
    container = ComposableNodeContainer(
            node_name='my_container',
            node_namespace='',
            package='rclcpp_components',
            node_executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='composition',
                    node_plugin='composition::Talker',
                    parameters=[{"delay": 500}],
                    node_name='fast_talker'),
                ComposableNode(
                    package='composition',
                    node_plugin='composition::Talker',
                    parameters=[{"delay": 5000}],
                    node_name='slow_talker'),
                ComposableNode(
                    package='composition',
                    node_plugin='composition::Listener',
                    node_name='listener')
            ],
            output='screen',
    )
```
Add this to `Talker::Talker` in `talker_component.cpp`:
```
  // Use a timer to schedule periodic message publishing.
  talk_speed_ = declare_parameter("delay", 1000);

  timer_ = create_wall_timer(std::chrono::duration<int, std::milli>(talk_speed_), std::bind(&Talker::on_timer, this));
```
In `on_timer()` add:
```
  pub_->publish(std::move(msg));
  std::this_thread::sleep_for(std::chrono::duration<int, std::milli>(talk_speed_ - 100));
  RCLCPP_INFO(this->get_logger(), "And slept");
  std::flush(std::cout);
```

#### Expected behavior
`fast_talker` should output every 0.5 seconds and `slow_talker` every 5 seconds; this is what a ROS1 launch file with nodelets would do.

#### Actual behavior
```
[component_container-1] [INFO] [fast_talker]: Publishing: 'Hello World: 1'
[component_container-1] [INFO] [fast_talker]: And slept
[component_container-1] [INFO] [listener]: I heard: [Hello World: 1]
[component_container-1] [INFO] [fast_talker]: Publishing: 'Hello World: 2'
[component_container-1] [INFO] [fast_talker]: And slept
...
[component_container-1] [INFO] [fast_talker]: Publishing: 'Hello World: 10'
[component_container-1] [INFO] [fast_talker]: And slept
[component_container-1] [INFO] [slow_talker]: Publishing: 'Hello World: 1'
[component_container-1] [INFO] [slow_talker]: And slept
[component_container-1] [INFO] [listener]: I heard: [Hello World: 10]
[component_container-1] [INFO] [fast_talker]: Publishing: 'Hello World: 11'
[component_container-1] [INFO] [fast_talker]: And slept
[component_container-1] [INFO] [slow_talker]: Publishing: 'Hello World: 2'
[component_container-1] [INFO] [slow_talker]: And slept
[component_container-1] [INFO] [listener]: I heard: [Hello World: 1]
[component_container-1] [INFO] [fast_talker]: Publishing: 'Hello World: 12'
[component_container-1] [INFO] [fast_talker]: And slept
[component_container-1] [INFO] [slow_talker]: Publishing: 'Hello World: 3'

```
#### Additional information
The `listener` is also stalled by the `sleep()`, but that is as expected. In a real example there would be no `sleep()`, but instead heavy processing. Our ROS1 code that we are moving over to ROS2 takes it as a given that multiple nodelets (even multiple instances of the same nodelet) within the same manager each run on separate threads.

hidmic, 2020-06-11:
@KenYN this is simply due to `launch_ros.actions.ComposableNodeContainer` using the single threaded [composable nodes' container](https://github.com/ros2/rclcpp/blob/master/rclcpp_components/src/component_container.cpp) instead of the [multithreaded one](https://github.com/ros2/rclcpp/blob/master/rclcpp_components/src/component_container_mt.cpp). 

I can imagine `launch_ros.actions.ComposableNodeContainer` taking an `execution_policy` argument to opt for single-threaded or multi-threaded execution. @ivanpauno @wjwwood for thoughts on that API.

In the meantime, would you be willing to contribute a patch?

KenYN, 2020-06-12:
@hidmic I've had a look at the source code and I'm not sure where I would start with a patch...

KenYN, 2020-06-17:
I've found that
```
ComposableNodeContainer(
            executable='component_container_mt',
             ...)
```
Does what I want. The performance is dreadful, however, but I'm not sure what the cause is, but that's a separate issue.

closed 2020-06-17 by KenYN

ivanpauno, 2020-06-17:
> Does what I want. The performance is dreadful, however, and I'm not sure what the cause is, but that's a separate issue.

The cause is probably the performance of the `MultiThreadedExecutor` itself.

KenYN, 2020-06-19:
I've seen the reports of that, but in the end the performance issue was a timing bug in my code that surfaced due to `ros1_bridge` overhead and other things...

ivanpauno, 2020-06-19:
> I've seen the reports of that, but in the end the performance issue was a timing bug in my code that surfaced due to `ros1_bridge` overhead and other things...

Thanks for clarifying @KenYN !!