ros2/launch_ros #171: LoadComposableNodes action can cause launch process to hang forever

jacobperron, 2020-08-14:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Any
- Version or commit hash:
  - Any
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Run this broken mock node component container:

```python
import time
import threading

from composition_interfaces.srv import LoadNode
import rclpy
import rclpy.context
import rclpy.executors
import rclpy.node

class MockComponentContainer(rclpy.node.Node):

    def __init__(self):
        # List of LoadNode requests received
        self.requests = []

        self._context = rclpy.context.Context()
        rclpy.init(context=self._context)

        super().__init__('mock_component_container', context=self._context)

        self.load_node_service = self.create_service(LoadNode, '~/_container/load_node', self.load_node_callback)

        self._executor = rclpy.executors.SingleThreadedExecutor(context=self._context)

        # Start spinning in a thread
        self._thread = threading.Thread(target=rclpy.spin, args=(self, self._executor), daemon=True)
        self._thread.start()

    def load_node_callback(self, request, response):
        print(f'Request received: {request}')
        self.requests.append(request)
        response.success = True
        response.full_node_name = f'{request.node_namespace}/{request.node_name}'
        response.unique_id = len(self.requests)
        # BROKEN HERE (forgot to return response)

    def ok(self):
        return rclpy.ok(context=self._context)

def main():
    container = MockComponentContainer()
    while container.ok():
        time.sleep(1)

if __name__ == '__main__':
    main()
```

Launch a LoadComposableNodes action pointing to the broken container:

```
import launch

from launch_ros.actions import LoadComposableNodes
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    return launch.LaunchDescription([
        LoadComposableNodes(
            target_container='/mock_component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='demo_nodes_cpp',
                    plugin='demo_nodes_cpp::Talker',
                )
            ]
        )
    ])
```

#### Expected behavior

LoadComposableNodes action fails to load the node gracefully.

#### Actual behavior

The launch process hangs forever. Even trying to send sigint or sigquit fails. We need to sigkill the launch process.

#### Additional information

After some debugging, it appears that we get stuck in this synchronous service call:

https://github.com/ros2/launch_ros/blob/91097d16e059f78fb9af511fd8ebee74af213e86/launch_ros/launch_ros/actions/load_composable_nodes.py#L103

And so the future being tracked by the launch context is never completed and launch waits indefinitely.

Perhaps there are two bugs here,

1. Launch should probably not wait indefinitely for all futures to complete during shutdown. Rather, it should probably timeout and cancel these hung futures.
2. We can fix the LoadComposableNodes action so that makes an asynchronous service call and monitors the launch shutdown state (similar to the [LifecycleNode action](https://github.com/ros2/launch_ros/blob/91097d16e059f78fb9af511fd8ebee74af213e86/launch_ros/launch_ros/actions/lifecycle_node.py#L116-L132)) .

jacobperron, 2020-08-14:
@wjwwood @ivanpauno What do you think about the first point I listed above?

> Launch should probably not wait indefinitely for all futures to complete during shutdown. Rather, it should probably timeout and cancel these hung futures.

If you agree, I can look into patching the launch service so that it has a timeout for waiting on futures during shutdown. Alternatively, we can at least add back a debug log telling the user that launch is still waiting on futures, but I don't think this is that useful, especially if we can't tell where the futures came from.

ivanpauno, 2020-08-14:
> If you agree, I can look into patching the launch service so that it has a timeout for waiting on futures during shutdown. Alternatively, we can at least add back a debug log telling the user that [...]

The problem with timeouts is defining how much is "too much".
I think it's reasonable to put some maximum, e.g.: 1min.
It will help avoiding shooting ourselves in the feet.

> Alternatively, we can at least add back a debug log telling the user that launch is still waiting on future

Adding back the debug logs is surely a good idea.

jacobperron, 2020-08-14:
The other option is to not even have a timeout; if launch is shutting down, just cancel all pending futures.

ivanpauno, 2020-08-14:
> The other option is to not even have a timeout; if launch is shutting down, just cancel all pending futures.

I think there is a reason why we're not doing that, I don't recall correctly.
Maybe @wjwwood or @hidmic remember it.

wjwwood, 2020-08-17:
> Launch should probably not wait indefinitely for all futures to complete during shutdown. Rather, it should probably timeout and cancel these hung futures.

I don't know off hand, but I'd say it would be better if they were not canceled and instead they could just shutdown cleanly. That being said having it error out at some point rather than hanging might be ok. But there might be cases where a long shutdown is needed, like you're waiting on a file to close or something.

jacobperron, 2020-08-17:
> But there might be cases where a long shutdown is needed, like you're waiting on a file to close or something.

That's a good point.

I'll fix the issue in `LoadComposableNodes` and leave the `launch` shutdown logic for now. If other users run into this, maybe we can add a launch option for a shutdown timeout (or consider letting the sigquit signal cancel any outstanding futures).

wjwwood, 2020-08-17:
> consider letting the sigquit signal cancel any outstanding futures

This seems reasonable to me.

commit 59c360fe (2020-08-18): Asynchronously wait for load node service response

Fixes #171

By asychronously waiting for the service response, we can monitor if launch is shutting down and abandon the request so we don't block the shutdown process.
PR #174 (2020-08-28): Asynchronously wait for load node service response
Fixes #171

By asychronously waiting for the service response, we can monitor if launch is shutting down and abandon the request so we don't block the shutdown process.
commit 066d8fe4 (2020-08-28): Asynchronously wait for load node service response (#174)

* Asynchronously wait for load node service response

Fixes #171

By asychronously waiting for the service response, we can monitor if launch is shutting down and abandon the request so we don't block the shutdown process.

* Add back debug log

It was accidentally removed.
commit 005dd9d6 (2020-11-19): Asynchronously wait for load node service response (#174)

* Asynchronously wait for load node service response

Fixes #171

By asychronously waiting for the service response, we can monitor if launch is shutting down and abandon the request so we don't block the shutdown process.

* Add back debug log

It was accidentally removed.
closed 2020-08-28 by jacobperron