ros2/launch_ros #181: 'remappings' is a confusing word. Possibly rename to simply "mapping"?

33thou, 2020-09-28:
I'm still getting familiar with the ROS ecosystem and community feedback mechanisms to feel free to point me in the right direction if this is the wrong place for this feedback.

I was going through the tutorial at https://index.ros.org/doc/ros2/Tutorials/Launch-Files/Creating-Launch-Files/

There is a parameter called 'remappings' to the `Node` object which seems incorrect. This parameter is simply a singular 'mapping' array of tuples.

clalancette, 2020-09-29:
> There is a parameter called 'remappings' to the `Node` object which seems to be incorrectly named. This parameter is simply a singular 'mapping' list of tuples.

The thing is, we call it `remapping` in a lot of places.  For instance, the design article explicitly uses that term: http://design.ros2.org/articles/static_remapping.html .

So I think for consistency, the term `remappings` in launch_ros makes the most sense.

33thou, 2020-09-29:
makes sense, thanks!

closed 2020-09-29 by 33thou