ros2/launch_ros #196: SetRemap does not work inside a launch include action

jacobperron, 2020-10-28:
## Feature request

#### Feature description

The SetRemap action was added in https://github.com/ros2/launch_ros/pull/158 to give users an easy way to apply remaps to multiple nodes at the same scope. However, if the SetRemap action is placed *inside* an include action, then the remappings are not applied.

For example, I want to remap the topic for nodes included in a launch file:

```xml
<launch>
    <include file="talker.launch.xml">
        <set_remap from="chatter" to="foo" />
    </include>
</launch>
```

But, in order to actually get it to work we have to put the `<set_remap>` action as a peer to `<include>`:

```xml
<launch>
    <set_remap from="chatter" to="foo" />
    <include file="talker.launch.xml">
    </include>
</launch>
```

This is less desirable because the remapping might leak to other nodes at include scope and we're forced to wrap everything in a `<group>`:

```xml
<launch>
    <group>
        <set_remap from="chatter" to="foo" />
        <include file="talker.launch.xml">
        </include>
    </group>
</launch>
```

I don't know if there's a technical reason for the current behavior or if it is simply an oversight.

jacobperron, 2020-10-28:
As an aside, I think it would be nice to just name the action `<remap>` (like in ROS 1). Maybe there's a good reason to not change the name though.

ivanpauno, 2020-11-09:
> For example, I want to remap the topic for nodes included in a launch file:

`include` doesn't take any subtag.

> I don't know if there's a technical reason for the current behavior or if it is simply an oversight.

The main reason is that `include` is defined in `launch` and not in `launch_ros`.
The only two subtags that make sense are `set_param` and `set_remap`.

I think that now that https://github.com/ros2/launch_ros/pull/198 and https://github.com/ros2/launch/pull/468 have landed this should at least not be confusing.

jacobperron, 2020-11-10:
> The main reason is that include is defined in `launch` and not in `launch_ros`.

That makes sense.

After looking at this again, I was mistakenly thinking that this was possible in ROS 1, but it is not. We accomplish this the same way in ROS 1's roslaunch, e.g.:

```xml
<launch>
    <remap from="chatter" to="foo" />
    <include file="talker.launch" />
</launch>
```

closed 2020-11-10 by jacobperron