ros2/launch_ros #217: "Fatal Python Error:Aborted" in test_composable_node_container using RTI Connext in Foxy Nightly

sloretz, 2021-02-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal
- Installation type:
  - Source
- Version or commit hash:
  - 7013bc3667c909a29e62f1b83d0ace628ad50a55>
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Unknown

#### Expected behavior

Tests pass

#### Actual behavior

It looks like the [cpython interpreter is receiving the SIGABRT signal](https://github.com/python/cpython/blob/7b3ab5921fa25ed8b97b6296f97c5c78aacf5447/Modules/faulthandler.c#L344).

#### Additional information

https://build.ros2.org/job/Fci__nightly-connext_ubuntu_focal_amd64/230/consoleFull#console-section-1935

<details><summary>Copy of output from job</summary>

```
03:18:40 Starting >>> test_launch_ros
03:18:41 ============================= test session starts ==============================
03:18:41 platform linux -- Python 3.8.5, pytest-6.2.2, py-1.10.0, pluggy-0.13.1
03:18:41 cachedir: /tmp/ws/build_isolated/test_launch_ros/.pytest_cache
03:18:41 rootdir: /tmp/ws/src/ros2/launch_ros/test_launch_ros
03:18:41 plugins: ament-copyright-0.9.6, ament-pep257-0.9.6, ament-flake8-0.9.6, ament-lint-0.9.6, rerunfailures-9.1.1, mock-1.10.4, colcon-core-0.6.1, cov-2.8.1
03:18:41 collected 100 items
03:18:41 
03:18:41 test/test_copyright.py .                                                 [  1%]
03:18:41 test/test_flake8.py .                                                    [  2%]
03:18:42 test/test_pep257.py .                                                    [  3%]
03:19:01 test/test_launch_ros/actions/test_composable_node_container.py F.F       [  6%]
03:19:01 test/test_launch_ros/actions/test_lifecycle_node.py ..                   [  8%]
03:19:14 Fatal Python error: Aborted
03:19:14 
03:19:14 Current thread 0x00007f8bca7fc700 (most recent call first):
03:19:14   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 586 in _wait_for_ready_callbacks
03:19:14   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 689 in wait_for_ready_callbacks
03:19:14   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 703 in spin_once
03:19:14   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/__init__.py", line 191 in spin
03:19:14   File "/usr/lib/python3.8/threading.py", line 870 in run
03:19:14   File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
03:19:14   File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap
03:19:14 
03:19:14 Thread 0x00007f8bc9ffb700 (most recent call first):
03:19:14   File "/usr/lib/python3.8/concurrent/futures/thread.py", line 78 in _worker
03:19:14   File "/usr/lib/python3.8/threading.py", line 870 in run
03:19:14   File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
03:19:14   File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap
03:19:14 
03:19:14 Thread 0x00007f8c105b8740 (most recent call first):
03:19:14   File "/usr/lib/python3.8/threading.py", line 1027 in _wait_for_tstate_lock
03:19:14   File "/usr/lib/python3.8/threading.py", line 1011 in join
03:19:14   File "/tmp/ws/src/ros2/launch_ros/test_launch_ros/test/test_launch_ros/actions/test_load_composable_nodes.py", line 81 in shutdown
03:19:14   File "/tmp/ws/src/ros2/launch_ros/test_launch_ros/test/test_launch_ros/actions/test_load_composable_nodes.py", line 120 in mock_component_container
03:19:14   File "/usr/local/lib/python3.8/dist-packages/_pytest/fixtures.py", line 941 in _teardown_yield_fixture
[... 81 lines cut ...]
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/manager.py", line 93 in _hookexec
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/hooks.py", line 286 in __call__
03:19:14   File "/usr/local/lib/python3.8/dist-packages/_pytest/main.py", line 348 in pytest_runtestloop
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/callers.py", line 187 in _multicall
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/manager.py", line 84 in <lambda>
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/manager.py", line 93 in _hookexec
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/hooks.py", line 286 in __call__
03:19:14   File "/usr/local/lib/python3.8/dist-packages/_pytest/main.py", line 323 in _main
03:19:14   File "/usr/local/lib/python3.8/dist-packages/_pytest/main.py", line 269 in wrap_session
03:19:14   File "/usr/local/lib/python3.8/dist-packages/_pytest/main.py", line 316 in pytest_cmdline_main
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/callers.py", line 187 in _multicall
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/manager.py", line 84 in <lambda>
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/manager.py", line 93 in _hookexec
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pluggy/hooks.py", line 286 in __call__
03:19:14   File "/usr/local/lib/python3.8/dist-packages/_pytest/config/__init__.py", line 162 in main
03:19:14   File "/usr/local/lib/python3.8/dist-packages/_pytest/config/__init__.py", line 185 in console_main
03:19:14   File "/usr/local/lib/python3.8/dist-packages/pytest/__main__.py", line 5 in <module>
03:19:14   File "/usr/lib/python3.8/runpy.py", line 87 in _run_code
03:19:14   File "/usr/lib/python3.8/runpy.py", line 194 in _run_module_as_main
03:19:14 ---
03:19:14 Failed   <<< test_launch_ros [34.3s, exited with code -6]
```
</details>

Linking to osrf/buildfarmer#145

chapulina, 2021-06-28:
[Another failure](https://build.ros2.org/view/Fci/job/Fci__nightly-cyclonedds_ubuntu_focal_amd64/286/consoleFull#console-section-1228) this week, this time with CycloneDDS.

The output seems to be the same as the one on the description, but I'll paste it here as reference:

<details><summary>Output</summary>
<p>

```
01:43:41 Starting >>> test_launch_ros
01:43:42 ============================= test session starts ==============================
01:43:42 platform linux -- Python 3.8.5, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
01:43:42 cachedir: /tmp/ws/build_isolated/test_launch_ros/.pytest_cache
01:43:42 rootdir: /tmp/ws/src/ros2/launch_ros/test_launch_ros
01:43:42 plugins: ament-copyright-0.9.6, ament-pep257-0.9.6, ament-flake8-0.9.6, ament-lint-0.9.6, rerunfailures-10.0, mock-1.10.4, colcon-core-0.6.1, cov-2.8.1
01:43:42 collected 103 items
01:43:42 
01:43:42 test/test_copyright.py .                                                 [  0%]
01:43:43 test/test_flake8.py .                                                    [  1%]
01:43:43 test/test_pep257.py .                                                    [  2%]
01:43:47 test/test_launch_ros/actions/test_composable_node_container.py ...       [  5%]
01:43:47 test/test_launch_ros/actions/test_lifecycle_node.py ..                   [  7%]
01:43:48 Fatal Python error: Aborted
01:43:48 
01:43:48 Current thread 0x00007fd3c0b7b700 (most recent call first):
01:43:48   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 586 in _wait_for_ready_callbacks
01:43:48   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 689 in wait_for_ready_callbacks
01:43:48   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 703 in spin_once
01:43:48   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/__init__.py", line 191 in spin
01:43:48   File "/usr/lib/python3.8/threading.py", line 870 in run
01:43:48   File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
01:43:48   File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap
01:43:48 
01:43:48 Thread 0x00007fd3a2ffd700 (most recent call first):
01:43:48   File "/usr/lib/python3.8/concurrent/futures/thread.py", line 78 in _worker
01:43:48   File "/usr/lib/python3.8/threading.py", line 870 in run
01:43:48   File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
01:43:48   File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap
01:43:48 
01:43:48 Thread 0x00007fd3c6d34740 (most recent call first):
01:43:48   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/context.py", line 120 in _logging_fini
01:43:48   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/context.py", line 88 in shutdown
01:43:48   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/utilities.py", line 58 in shutdown
01:43:48   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/__init__.py", line 101 in shutdown
01:43:48   File "/tmp/ws/src/ros2/launch_ros/test_launch_ros/test/test_launch_ros/actions/test_load_composable_nodes.py", line 81 in shutdown
01:43:48   File "/tmp/ws/src/ros2/launch_ros/test_launch_ros/test/test_launch_ros/actions/test_load_composable_nodes.py", line 120 in mock_component_container
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/fixtures.py", line 941 in _teardown_yield_fixture
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/fixtures.py", line 1027 in finish
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 395 in _callfinalizers
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 405 in _teardown_with_finalization
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 387 in _pop_and_teardown
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 427 in _teardown_towards
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 419 in teardown_exact
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 175 in pytest_runtest_teardown
01:43:48   File "/usr/local/lib/python3.8/dist-packages/pluggy/callers.py", line 187 in _multicall
01:43:48   File "/usr/local/lib/python3.8/dist-packages/pluggy/manager.py", line 84 in <lambda>
01:43:48   File "/usr/local/lib/python3.8/dist-packages/pluggy/manager.py", line 93 in _hookexec
01:43:48   File "/usr/local/lib/python3.8/dist-packages/pluggy/hooks.py", line 286 in __call__
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 255 in <lambda>
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 311 in from_call
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 254 in call_runtest_hook
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 215 in call_and_report
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 127 in runtestprotocol
01:43:48   File "/usr/local/lib/python3.8/dist-packages/_pytest/runner.py", line 109 in pytest_runtest_protocol
01:43:48   File "/usr/local/lib/python3.8/dist-packages/pluggy/callers.py", line 187 in _multicall
01:43:48   File "/usr/local/lib/python3.8/dist-packages/pluggy/manager.py", line 84 in <lambda>
[... cut ...]
01:43:49   File "/usr/local/lib/python3.8/dist-packages/_pytest/config/__init__.py", line 185 in console_main
01:43:49   File "/usr/local/lib/python3.8/dist-packages/pytest/__main__.py", line 5 in <module>
01:43:49   File "/usr/lib/python3.8/runpy.py", line 87 in _run_code
01:43:49   File "/usr/lib/python3.8/runpy.py", line 194 in _run_module_as_main
01:43:49 ---
01:43:49 Failed   <<< test_launch_ros [7.72s, exited with code -6]
```

</p>
</details>

https://github.com/osrf/buildfarmer/issues/207

chapulina, 2021-06-29:
I ran this a few times locally on Ubuntu Focal with CycloneDDS and I can't reproduce the failure.

Looking at the various logs for this issue, it seems that the SIGABRT always happens during shutdown.

clalancette, 2021-07-01:
I've launch two testing jobs for this on the buildfarm:

* https://build.ros2.org/view/Fci/job/Fci__overlay_ubuntu_focal_amd64/40 is setup exactly the same as the failing nightly to make sure we can reproduce
* https://build.ros2.org/view/Fci/job/Fci__overlay_ubuntu_focal_amd64/41 is setup the same as 40, except that the recent change to osrf_pycommon is reverted.  This is to see if that change is the culprit

chapulina, 2021-07-02:
> I've launch two testing jobs for this on the buildfarm:

Both of them failed in a different way from this issue :upside_down_face: 

```
17:43:56 INFO: The total number of tests for the threshold 'Failed Tests' exceeds the specified "unstable threshold" value.
```

And I don't see `test_launch_ros` being tested on either of them.

j-rivero, 2021-07-12:
> Both of them failed in a different way from this issue upside_down_face

Both of them finished with a SIGINT. I'm pretty sure that the problem is that it reached the timeout set at 240 minutes, exactly 4 hours. The `Fci__` jobs seems to be set to 240 minutes of timeout while the nightly builds are configured to 300 minutes.

sloretz, 2021-09-20:
Saw this issue again: https://build.ros2.org/view/Fci/job/Fci__nightly-connext_ubuntu_focal_amd64/312

sloretz, 2021-09-24:
grep'ing through the Connext Nightly job, I see a lot of "Fatal Python error: Aborted" in tests other than launch_ros. Maybe this is an issue from something lower level - rclpy or rosidl_python?

Blast545, 2021-11-03:
This issue was addressed with this backport: [launch_ros#279](https://github.com/ros2/launch_ros/pull/279).

Waited half a month to be sure it didn't happen again, for now I'll cose this issue.

closed 2021-11-03 by Blast545