ros2/launch_ros #220: push-ros-namespace is not working for LifecycleNode if namespace is empty

kenji-miyake, 2021-03-14:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.11.1
- DDS implementation:
  - default

#### Steps to reproduce issue

```xml
// test.launch.xml
<launch>
  <group>
    <push-ros-namespace namespace="test"/>
    <include file="$(find-pkg-share lifecycle)/launch/lifecycle_demo.launch.py" >
    </include>
  </group>
</launch>
```

```sh
$ ros2 launch ./test.launch.xml
$ ros2 node list
/launch_ros_2173028
/lc_talker
/test/lc_client
/test/lc_listener
```

#### Expected behavior

Lifecycle nodes will be also namespaced.
In this case, `/lc_talker` should be `/test/lc_talker`.

#### Actual behavior

No namespace is given to the lifecycle node.

#### Additional information

This is because [if namespace is `''` or `None`, it's set to `/`](https://github.com/ros2/launch_ros/blob/4e74f5a60e023b8d810f17df463692a584cd8575/launch_ros/launch_ros/actions/lifecycle_node.py#L93-L94).
[lifecycle_demo.launch.py](https://github.com/ros2/demos/blob/b0c1389ae61172788cccf4eb4e34b3b84c408a9d/lifecycle/launch/lifecycle_demo.launch.py#L22-L23) uses `''` for namespace.

This behavior seems to be introduces in https://github.com/ros2/launch_ros/pull/179.

I found a workaround, using `TextSubstitution` can avoid this problem.

```py
// lifecycle_demo_workaround.launch.py
from launch import LaunchDescription
from launch.substitutions import TextSubstitution
from launch_ros.actions import LifecycleNode
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        LifecycleNode(package='lifecycle', executable='lifecycle_talker',
                      name='lc_talker', namespace=TextSubstitution(text=''), output='screen'),
        Node(package='lifecycle', executable='lifecycle_listener', output='screen'),
        Node(package='lifecycle', executable='lifecycle_service_client', output='screen')
    ])
```

```xml
// workaround.launch.xml
<launch>
  <group>
    <push-ros-namespace namespace="test"/>
    <include file="./lifecycle_demo_workaround.launch.py" >
    </include>
  </group>
</launch>
```

```sh
$ ros2 launch ./workaround.launch.xml
$ ros2 node list
/launch_ros_2173789
/test/lc_client
/test/lc_listener
/test/lc_talker
```

jacobperron, 2021-03-26:
Oops, looks like #179 wasn't backwards compatible after all.

Thanks for the detailed report!

I think this is the correct way to patch it: https://github.com/ros2/launch_ros/pull/222

@kenji-miyake Could you give that PR a try?

kenji-miyake, 2021-03-26:
@jacobperron Thank you for checking this! Okay, I'll try it. 😄

kenji-miyake, 2021-03-28:
@jacobperron I'm sorry to be late. I've checked the PR and it worked for my use case!
I tried some patterns:
1. `test.launch.xml` (as written in the description)
2. Disable `push-ros-namespace` in `test.launch.xml`
3. Add `namespace` to `lifecycle_demo.launch.py`

PR #222 (2021-03-31): Fix namespace with LifecycleNode
Fixes #220.
Regression since #179.

The expectation prior to #179 was that providing no namespace argument
to LifecycleNode would result in an empty, relative namespace.
This change fixes the code to meet that expectation.
commit f9bb0cab (2021-03-31): Fix namespace with LifecycleNode (#222)

Fixes #220.
Regression since #179.

The expectation prior to #179 was that providing no namespace argument
to LifecycleNode would result in an empty, relative namespace.
This change fixes the code to meet that expectation.
closed 2021-03-31 by jacobperron