ros2/launch_ros #221: [ros2launch] [cli] ros2 launch fails with absolute path and argument

rob-clarke, 2021-03-18:
## Bug report

**Required Info:**

- Operating System:
ros:foxy docker image on Ubuntu 18.04 (Likely affects all)
- Installation type:
binary in docker image (Likely affects all)
- Version or commit hash:
Version reported in ros2launch package.xml is 0.11.1
- DDS implementation:
Whichever is default in docker image (Likely affects all)
- Client library (if applicable):
Not applicable to issue

#### Steps to reproduce issue

##### Setup
Spawn a container and enter shell:
```
docker run -it ros:foxy bash
```
In the image, install demo_nodes_cpp to provide a target launch file:
```
apt update && apt install ros-foxy-demo-nodes-cpp
```

##### Issue causing variant
Test absolute `'single file'` launching variant with argument:
```
ros2 launch /opt/ros/foxy/share/demo_nodes_cpp/launch/services/add_two_ints_async.launch.py example_arg:="foobar"
```

##### Functional variants
Test `'pkg file'` launching variant:
```
ros2 launch demo_nodes_cpp add_two_ints_async.launch.py
```
Test `'pkg file'` launching variant, with argument:
```
ros2 launch demo_nodes_cpp add_two_ints_async.launch.py example_arg:="foobar"
```
Test absolute `'single file'` launching variant:
```
ros2 launch /opt/ros/foxy/share/demo_nodes_cpp/launch/services/add_two_ints_async.launch.py
```

#### Expected behavior

Successfully runs the launch file for all variants.

#### Actual behavior

Absolute `'single file'` with argument fails with error:
```
file 'example_arg:=foobar' was not found in the share directory of package '/opt/ros/foxy/share/demo_nodes_cpp/launch/services/add_two_ints_async.launch.py' which is at '/opt/ros/foxy/share/demo_nodes_cpp/launch/services/add_two_ints_async.launch.py'
```

#### Additional information

This is due to behaviour of `ament_index_python` when passed an absolute path, see [PR#69 on ament_index](https://github.com/ament/ament_index/pull/69) for more details. In brief, if a valid absolute path was passed to `get_package_prefix()` as the package name, a `PackageNotFoundError` was __not__ raised.

I had begun to put together a fix for this but got lost trying to implement a set of tests for the CLI. Lack of CLI tests is potentially worthy of a separate issue.

hidmic, 2021-03-18:
Thanks for reporting @rob-clarke ! A fix is most welcome. Handling `get_package_prefix()` errors should be enough.

rob-clarke, 2021-03-19:
If the PR on ament index is incorporated. I'm happy to expand the current flow to include the new `InvalidResourceNameError`. Is there any way of versioning the dependencies?

hidmic, 2021-03-19:
There is, but it is usually implied by the ROS distribution. Both `ament_index` and `ros2launch` fixes will apply to Rolling (i.e. the rolling distribution). To have them in Foxy they'll need to get backported. Since these can be deemed bug-fixes, that door remains open I believe.

rob-clarke, 2021-06-13:
Upstream PR has been merged so root issue is dealt with.

I've created a couple of PRs here that sidestep the issue by defaulting to `'single file'` mode and only switching to `'pkg file'` if the specified file does not exist. Confusing error messages may still occur but successfully completes all of the tests outlined above. One is against foxy, while one is against master.

Formalising the tests would be a good next step. Raised an issue in #246

rob-clarke, 2021-06-15:
@hidmic / @clalancette Could you take a look at the PRs?
#230 against master and
#245 against foxy

closed 2021-07-16 by hidmic

JakubHazik, 2022-02-21:
This bug is still reproducable on the newes `galactic` distribution. Do you plan to fix it?

jacobperron, 2022-02-23:
Good catch. I think we should make it consistent between Foxy and Rolling. Here's a backport for Galactic: https://github.com/ros2/launch_ros/pull/296

PR #230 (2021-07-16): Simplify logic to fix absolute paths
Simplifies the logic to fix behaviour for absolute paths.

Fixes #221 

There is a caveat for the `'pkg file'` form: if a _file_ with the same name as the target package exists in the current working directory, this will attempt to open that file as a launch file. This breaks from existing behaviour but should be a rare occurrence.
PR #245 (2021-09-30): Foxy: Simplify logic for mode switch
Backport of #230 for foxy
Fixes #221