ros2/launch_ros #228: Could not finish process when launching a component.

k0suke-murakami, 2021-04-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 
- Installation type:
  - binaries 
- Version or commit hash:
  - `$ apt-cache policy ros-foxy-launch-ros` 
  0.11.1-1focal.20210225.220443 
- DDS implementation:
  - cyclone_dds 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

1. Clone demo repository
   `$ git clone git@github.com:ros2/demos.git`
2. Checkout to foxy
   `$ cd demos`
   `$ git checkout foxy`
 3. Insert sleep code in  `talk_component.cpp`
```
Talker::Talker(const rclcpp::NodeOptions & options)
: Node("talker", options), count_(0)
{
  // Create a publisher of "std_mgs/String" messages on the "chatter" topic.
  pub_ = create_publisher<std_msgs::msg::String>("chatter", 10);

  std::cerr << "sleep start" << std::endl;
  rclcpp::sleep_for(std::chrono::seconds(10));
  std::cerr << "sleep end" << std::endl;

  // Use a timer to schedule periodic message publishing.
  timer_ = create_wall_timer(1s, std::bind(&Talker::on_timer, this));
}

```
4. Build & Run
`$ ros2 launch composition composition_demo.launch.py`

5. Push `Ctrl + c` to finish the process soon after recognizing `sleep start` in the console.

#### Expected behavior
Finish the process after pressing `Ctrl + c`

#### Actual behavior
Could not finish the process even after pressing `Ctrl + c`.
The console says `C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...`.

#### Additional information

----
>

adamdbrw, 2021-04-20:
Confirmed that it happens with Foxy.

With Rolling, however:

```
[component_container-1] [INFO] [1618911813.423608225] [my_container]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[component_container-1] [INFO] [1618911813.423694366] [my_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[component_container-1] sleep start
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/my_container/_container/load_node' service response, due to shutdown.
[component_container-1] [INFO] [1618911814.923379285] [rclcpp]: signal_handler(signal_value=2)
[component_container-1] sleep end
[ERROR] [component_container-1]: process has died [pid 285223, exit code -11, cmd '/opt/ros/rolling/lib/rclcpp_components/component_container --ros-args -r __node:=my_container -r __ns:=/'].
```

note that it states "Abandoning wait" so the case seems to be handled.

If I add if `(rclcpp::ok())` check before creating the timer it instead misbehaves a bit on close:
```
[component_container-1] [INFO] [1618912111.413347351] [my_container]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[component_container-1] [INFO] [1618912111.413365359] [my_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[component_container-1] sleep start
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/my_container/_container/load_node' service response, due to shutdown.
[component_container-1] [INFO] [1618912112.494379736] [rclcpp]: signal_handler(signal_value=2)
[component_container-1] sleep end
[component_container-1] [ERROR] [1618912112.494487100] [rclcpp]: failed to finalize rcl node options: error not set
[component_container-1] free(): invalid pointer
[ERROR] [component_container-1]: process has died [pid 286593, exit code -6, cmd '/opt/ros/rolling/lib/rclcpp_components/component_container --ros-args -r __node:=my_container -r __ns:=/'].
```

I would say it should terminate more gracefully, but it is a definitely a different issue on Rolling than on Foxy.

hidmic, 2021-04-20:
This doesn't look like an issue with `launch_ros`. Hangs on Foxy indicate an issue with `launch` signal handling, which has changed a bit in Rolling (for the better I hope, and it seems like it in @adamdbrw repro). The rest is [`ComponentManager::OnLoadNode`](https://github.com/ros2/rclcpp/blob/b8df9347a10bff383fdc8890c5b1bfdf16430d8f/rclcpp_components/src/component_manager.cpp#L125) misbehaving or something else in `rclcpp` not handling exceptions during service request processing. CC @ivanpauno.

adamdbrw, 2021-04-23:
I though this might be solved by https://github.com/ros2/launch/pull/500 and https://github.com/ros2/launch_ros/pull/225/ which were only recently backported to Foxy, but it is not the case:
Most up to date versions which contain these PRs, 0.11.2 launch-ros and 0.10.5 launch, still have the issue.

adamdbrw, 2021-04-23:
Found it! It is enough to apply https://github.com/ros2/launch_ros/pull/174 to solve the issue. I saw that the issue was planned for backports as far back as Foxy patch 4, but never made it. Any insights why?
The original issue: https://github.com/ros2/launch_ros/issues/171

ivanpauno, 2021-08-09:
A backport for Foxy was done in https://github.com/ros2/launch_ros/pull/240, did that solve this issue @adamdbrw ?

adamdbrw, 2021-08-09:
Yes it did, we can close it.

closed 2021-08-09 by ivanpauno