ros2/launch_ros #233: Parameter from YAML file cannot be overridden

felixdivo, 2021-05-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binary via `apt`
- Version or commit hash:
  - Latest available Foxy @ `0.10.5-1focal.20210423.000156`
- DDS implementation:
  - default
- Client library (if applicable):
  - not relevant (`rclpy`)

#### Steps to reproduce issue

1. Create `example_launch.py`:

```python
#!/usr/bin/env python3

from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description() -> LaunchDescription:
    return LaunchDescription(
        [
            Node(
                package="turtlesim",
                executable="turtlesim_node",
                name="turtlesim_node",
                parameters=[
                    {"background_b": 200},  # should have no effect
                    "parameters.yaml",  # should override the default value
                    {"background_b": 200},  # should override the value 5 from the previous line
                ],
            ),
        ]
    )
```

2. Create `parameters.yaml`:

```yaml
turtlesim_node:
  ros__parameters:

    background_b: 5
```

3. Start with `ros2 launch example_launch.py`.

4. Check the param in another terminal: `ros2 param get turtlesim_node background_b`.

#### Expected behavior

Output `Integer value is: 200`.

#### Actual behavior

Output `Integer value is: 5`.

#### Additional information

[The documentation states that parameters declared later shall override prior ones.](https://github.com/ros2/launch_ros/blob/46bf3de8c995824784b57dd5e9b64e8872a40832/launch_ros/launch_ros/actions/node.py#L162-L176)

hidmic, 2021-05-03:
TL; DR check if specifying the node namespace makes it work.

PS: this should probably be a question in https://answers.ros.org/

I suspect the problem here is wildcard matching precedence in target node names for parameter overrides. In your parameter file, the target node for the parameter override is fully specified. Neither of the two `dict` instances you provide along that file have that information, and `launch_ros` doesn't know what the fully qualified node name for this `Node` is (as you provide a node name, but not a node namespace). Therefore, [it falls back to a wildcard](https://github.com/ros2/launch_ros/blob/8f7bfc5f7d2320c3c2003afa101683df42d6bda3/launch_ros/launch_ros/actions/node.py#L322). Alas, for both `rclpy` and `rclcpp`, parameter overrides for fully specified node names always take precedence over parameter overrides with wildcards (e.g. see [here](https://github.com/ros2/rclcpp/blob/1037822a63330495dcf0de5e8f20544375a5f116/rclcpp/src/rclcpp/detail/resolve_parameter_overrides.cpp#L56-L66)). 

If you fully specify your `turtlesim` node name and namespace, you _should_ get the effect you want.

felixdivo, 2021-05-04:
@hidmic Thanks for your swift response! I was convinced that this cannot be intended behavior and therefore posted about it here.

Indeed, once I add `namespace="/"` to the launch file it works fine. However, doesn't the namespace argument default to `/` anyways? Whether I use it or not, I get the same structure:

```bash
$ ros2 node list
/turtlesim_node
```

So shouldn't the empty namespace be considered equal to the `/` namespace and therefore result in parameters being treated equally?

Anyways, the documentation isn't too clear about this. At least I did not find any hint that wildcard overrides can cause "unexpected" beahviour.

hidmic, 2021-05-04:
> However, doesn't the namespace argument default to `/` anyways?

Not necessarily, no. When instantiating an [`rclcpp::Node`](https://github.com/ros2/rclcpp/blob/c15eb1eaac2e05b507164a557cda5ff11d49c813/rclcpp/include/rclcpp/node.hpp#L100-L104), some code might set its namespace to something else. It'd unfortunate if `launch_ros` forced a remapping to `/` by default.

> Anyways, the documentation isn't too clear about this. At least I did not find any hint that wildcard overrides can cause "unexpected" beahviour.

Agreed. Documentation contributions are super welcome.

PR #234 (2021-05-06): Better document parameter handling in Node
Closes #233.

cc @hidmic
closed 2021-05-06 by jacobperron