ros2/launch_ros #236: Support LifecycleNode and EventHandler in frontend

kenji-miyake, 2021-05-07:
## Feature request

#### Feature description

It's useful to support LifecycleNode in frontend(xml) as well.

As for LifecycleNode itself, if we only use simple launch files as below, it can be done by just exposing like this https://github.com/kenji-miyake/launch_ros/commit/8daecd810d65ebdea6df32ab145137faa343a80d.

```xml
<launch>
  <lifecycle_node pkg="lifecycle" exec="lifecycle_talker" name="lc_talker" namespace=""/>
  <node pkg="lifecycle" exec="lifecycle_listener" />
  <node pkg="lifecycle" exec="lifecycle_service_client" />
</launch>
```

However, I believe it's not enough useful because we usually would like to write state transitions as well like [this](https://github.com/autowarefoundation/ros2_socketcan/blob/a1f6cbf65e899deab01b0ccbc9084872580f9ae6/launch/socket_can_receiver.launch.py).

So I think it's better to support RegisterEventHandler and EventHandlers.

#### Implementation considerations

I think there are at least two problems.

1. How to expose EventHandlers?
Is it okay to add `parse_event_handler` and `instantiate_event_handler` as Action class does?

2. How to refer Python Object from frontend?
For example, `OnStateTransition` requires `LifecycleNode` object but I think it can't be given in frontend easily.
Probably this is what @ivanpauno mentioned here https://answers.ros.org/question/333404/launch-composed-nodes-using-the-launch-xml-frontend/, right?
> The topic of having references (i.e.: being able to identify) another action in the XML/YAML frontend is an interesting feature currently not supported. It would be useful, for example, to support event handlers.

So I guess some mechanism for finding Object from all entities.

Does anybody have any idea about this?
If somebody guides me, I'll try to work on this task!

clalancette, 2021-05-27:
@jacobperron To provide some advice on how to move forward with this.

jacobperron, 2021-05-27:
@kenji-miyake Certainly, what you have is a good start (kenji-miyake@8daecd8). I would accept that in a pull request on its own.

Regarding exposing events and event handlers (etc) in the front-end, I suggest we open up a separate issue in [launch](https://github.com/ros2/launch). I imagine it should be easy to expose the [RegisterEventHandler](https://github.com/ros2/launch/blob/3a5c9df2e384271637ac029f48f020b7916498d6/launch/launch/actions/register_event_handler.py#L28) action. Then we need a way to parse `EventHandlers`, which I would guess should be fairly easy except when we need to reference other launch actions (e.g. referring to the Python objects as you mention). In these cases, maybe we could limit functionality such that launch actions must be referenced by name (ie. a string). I haven't thought about it much, but if you'd like to continue the discussion it would be great if you opened up a ticket over in [launch](https://github.com/ros2/launch).

kenji-miyake, 2021-05-30:
Thank you!
I can't make time so much recently, but I'll try to work on this task, so please wait for that.
Or somebody else would like to proceed more quickly, then I'd appreciate it if they could open issues and implement a prototype.

esteve, 2022-06-28:
@kenji-miyake I can work on this. What is missing in your branch? How do you think it should continue? Thanks.

esteve, 2022-10-04:
> @kenji-miyake Certainly, what you have is a good start ([kenji-miyake/launch_ros@8daecd8](https://github.com/kenji-miyake/launch_ros/commit/8daecd8)). I would accept that in a pull request on its own.

@jacobperron I've created a PR based on @kenji-miyake 's work at https://github.com/ros2/launch_ros/pull/327, I'll work on the remaining tasks in separate PRs

esteve, 2022-10-04:
@jacobperron I've created https://github.com/ros2/launch/issues/658 to continue the discussion there about the RegisterEventHandler action

closed 2025-10-26 by christophebedard