ros2/launch_ros #239: ROS Timer Action

jacobperron, 2021-05-19:
## Feature request

#### Feature description

Similar to the [TimerAction](https://github.com/ros2/launch/blob/3a5c9df2e384271637ac029f48f020b7916498d6/launch/launch/actions/timer_action.py), it would be great to have a timer action that was aware of ROS time (ie. rather than wall clock, it could use ROS time as defined [here](https://design.ros2.org/articles/clock_and_time.html)).

#### Implementation considerations

I imagine usage would look almost identical to `TimerAction`, e.g.

```python
RosTimerAction(
    period=5.0,
    actions=[
        # some list of actions to execute every 5 seconds ...
    ],
    cancel_on_shutdown=True,
)
```

If we ensure the ROS node in the launch context is using ROS time, then this may be as easy as registering a ROS timer.

rebecca-butler, 2021-06-10:
I've changed the usage a bit:
```
RosTimerAction(
    period=5.0,
    actions=[
        # some list of actions to execute every 5 seconds ...
    ],
    use_sim_time=True,
)
```

` use_sim_time` must be set if the user wants to use sim time, otherwise the timer will use system time. This seems cleaner than requiring an extra action to set the use_sim_time parameter. I also removed `cancel_on_shutdown` argument because the Shutdown event automatically destroys ROS timers.

ivanpauno, 2021-06-10:
> Similar to the TimerAction, it would be great to have a timer action that was aware of ROS time (ie. rather than wall clock, it could use ROS time as defined here).

I have a hard time imagining great use cases for this, but I don't mind supporting it.

PR #244 (2021-06-15): ROS Timer Action
Closes #239
closed 2021-06-15 by rebecca-butler