ros2/launch_ros #255: Declare dependencies on launch_{xml,yaml} parsers to make sure they're available

christophebedard, 2021-07-14:
## Feature request

#### Feature description

In some cases, if a package uses/runs launch files through `ros2launch` or `launch_testing` and nothing declares a dependency on `launch_xml`/`launch_yaml`, then the XML/YAML parsers won't be available. This causes weird failures.

Some example packages do declare an `<exec_depend>` or `<test_depend>` on one of the parser packages (e.g. https://github.com/ros2/demos/blob/3af19c93c7abd9a83caca47de0780e941faeb22b/demo_nodes_cpp/package.xml#L26), but in general this is a bit unexpected.

See discussion under https://github.com/ros2/launch/pull/516#issuecomment-879445998

#### Implementation considerations

Simply declare `<exec_depend>`s on `launch_xml` and `launch_yaml` in `ros2launch` and `launch_testing` (note that the latter is not in this repo). This should cover most cases.

christophebedard, 2021-07-14:
cc @hidmic want me to do it?

hidmic, 2021-07-14:
That'd be great @christophebedard. I'll make it official.

closed 2021-11-12 by jacobperron