ros2/launch_ros #257: Selective application of launch-prefix from CLI

camm73, 2021-07-19:
## Feature request

#### Feature description
ros2/launch_ros#254 adds the `--launch-prefix` flag to provide a means of prepending the same prefix (e.g. `xterm -e valgrind`) to each executable command in the launch file. For large launch files this launches many instances of the desired prefix program even if the user may only require it for certain executables. Currently selectively applying prefixes can be achieved by editing the launch file and directly specifying the applied prefix for a given executable; however, this is a slow process if the prefix is frequently changed or the developer only wishes to use a prefix one time.

To provide a means of filtering which executables the `--launch-prefix` value is applied to, I propose that we add a `--launch-prefix-filter` flag (or something to this effect) which allows the user to provide a list of executable names for which the prefix should be applied. When used alone `--launch-prefix` would apply the prefix to all executables, and when used in conjunction with the new `--launch-prefix-filter` flag, the prefix would only be applied to matching executables.

The goal of these two flags is to provide a faster means of debugging multiple, specific executables without requiring the launch file to be edited.

#### Implementation considerations
For the `launch_ros` repo:

- Add `--launch-prefix-filter` argument which accepts at least 1 executable name
- Pass list of executable names as a `/` separated list via `service.context.launch_configurations` (this is because Unix & Windows filenames cannot contain `/`)

For the `launch` repo (Assuming ros2/launch_ros#254 provides the prefix via `service.context.launch_configurations`):

- Modify `ExecuteProcess.__expand_substitutions()` such that when a launch prefix filter is available in the launch configuration, the prefix is only prepended to the command if its executable name matches an executable name that was provided via the launch configuration

The `name` property within `ExecuteProcess` is purposely not used so that the behavior is the same between `Node` and `ExecuteProcess`. `Node` sets the `ExecuteProcess` name to the executable name by default, but when the user creates their own `ExecuteProcess` in a launch file, they are able to provide their own `name`, overriding it from being set to the executable name. This could cause an issue where user-named `ExecuteProcess` elements wouldn't be properly selected by the filter, so it's executable name should be extracted directly from the `cmd` property instead.

hidmic, 2021-08-06:
> Pass list of executable names as a `/` separated list via `service.context.launch_configurations` (this is because Unix & Windows filenames cannot contain `/`)

I like the idea but this bit is unusual. How about using regex instead?

camm73, 2021-08-09:
> I like the idea but this bit is unusual. How about using regex instead?

Yeah I agree—that seems like a cleaner implementation.

PR #261 (2021-09-03): Add regex filter for selective launch-prefix application
Resolves #257 

This PR shouldn't be merged until #254 and https://github.com/ros2/launch/pull/522 have been accepted and merged.

This PR adds the `--launch-prefix-filter` argument discussed in #257 which allows the user to provide a regex pattern that can be used for selectively applying the command prefix (provided via the `--launch-prefix` argument) to executables which match the pattern. This will allow an easier debugging experience without having to edit the launch file.
closed 2021-09-03 by hidmic