ros2/launch_ros #258: consider porting `RewrittenYaml` from Nav2 to allow param overrides to YAML from launch

adamsj-ros, 2021-07-26:
## Feature request
Consider porting `RewrittenYaml` [from Nav2](https://github.com/ros-planning/navigation2/pull/771/files#diff-ddc85dea3844fd855756b5c6040387f0c9bb0739ddf54eab519191d7021b9aefR19).

#### Feature description

The use case is using YAMLs for default params but allowing to override individual params from launch. The overwrites could be extended to an override yaml. 

```
launch_ros.actions.Node(
        package='nodes',
        executable='node_exe',
        parameters=[
            path_to_yaml,
            # overwrite parameters from yaml here
            {"param_name": value}
        ]
```
This seems to almost work but with the limitation that the params specified in the YAML must be for all nodes. This is not ideal in the event params have the same name between nodes and you wish to distinguish. 

#### Implementation considerations

[This](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/blob/master/src/tools/joystick_vehicle_interface_nodes/launch/joystick_vehicle_interface_node.launch.py#L64) is a pretty nice way to override individual parameters in a YAML during launch. However, it only seems to work if the YAML file is set to [for all nodes](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/src/tools/joystick_vehicle_interface_nodes/param/logitech_f310.default.param.yaml#L1).

An alternative is Nav2's `RewrittenYaml`[[link](https://github.com/ros-planning/navigation2/pull/771/files#diff-ddc85dea3844fd855756b5c6040387f0c9bb0739ddf54eab519191d7021b9aefR19)] feature but that requires to clone/install all of Nav2.

SteveMacenski, 2021-11-29:
I'm happy to donate any and all work from Nav2 that would have more general use across the community. I'd love if this was here!

hidmic, 2021-12-08:
> This seems to almost work but with the limitation that the params specified in the YAML must be for all nodes.

Right, because `Node` was not written with multi-node processes in mind. 

#215 will indirectly address this issue by generalizing process execution via `launch`. It needs work still though.