ros2/launch_ros #272: Feature request (``launch_testing``): Check if a topic is publishing messages, with a timeout

adityapande-1995, 2021-09-30:
## Feature request

#### Feature description
It would be nice to have a way to check if a topic is publishing messages (with a timeout). 
THis suggestion came up in https://github.com/ros2/launch_ros/pull/263

#### Implementation considerations
One way to do that can be :
```
from threading import Event, Thread
import unittest

import launch
import launch.actions
import launch_ros.actions
import launch_testing.actions
import pytest
import rclpy
from rclpy.node import Node
from std_msgs.msg import String

@pytest.mark.launch_test
def generate_test_description():
    return launch.LaunchDescription([
        launch_ros.actions.Node(
            package='demo_nodes_cpp',
            executable='talker',
            name='demo_node_1',
        ),

        launch_testing.actions.ReadyToTest()
    ])

class TestFixture(unittest.TestCase):

    def test_check_if_msgs_published(self, proc_output):
        rclpy.init()
        node = MakeTestNode('test_node')
        node.start_subscriber()
        msgs_received_flag = node.msg_event_object.wait(timeout=5.0)
        assert msgs_received_flag, 'Did not receive msgs !'
        rclpy.shutdown()

class MakeTestNode(Node):

    def __init__(self, name='test_node'):
        super().__init__(name)
        self.msg_event_object = Event()

    def start_subscriber(self):
        # Create a subscriber
        self.subscription = self.create_subscription(
            String,
            'chatter',
            self.subscriber_callback,
            10
        )

        # Add a spin thread
        self.ros_spin_thread = Thread(target=lambda node: rclpy.spin(node), args=(self,))
        self.ros_spin_thread.start()

    def subscriber_callback(self, data):
        self.msg_event_object.set()

```

jacobperron, 2021-09-30:
IMO, ideally syntax would look like this:
```
def test_foo(self):
    wait_for_topics = WaitForTopics(['foo', 'bar', 'baz'])
    assert wait_for_topics.wait(timeout=5.0)
    ...
```

However, we need an instance to a ROS node (and context). Maybe we can use the existing [`ROSAdapter`](https://github.com/ros2/launch_ros/blob/999b208a05998a83378f7ba273b8146ea3eb09d9/launch_ros/launch_ros/ros_adapters.py#L29) class, e.g.:

```
class WaitForTopics:
   def __init__(self, *, topics):
       self.__ros_adapter = ROSAdapter()
       self.__ros_adapter.start()

    def __del__(self):
       self.__ros_adapter.shutdown()
```

@ivanpauno @wjwwood, thoughts?

adityapande-1995, 2021-10-04:
Relevant PR : https://github.com/ros2/launch_ros/pull/274