ros2/launch_ros #285: :man_farmer: `test_launch_ros` flaky Windows hangs

Blast545, 2021-11-05:
## Bug report

**Required Info:**

- Operating System:
  - Windows
- Installation type:
  - Source
- Version or commit hash:
  - `master`

#### Steps to reproduce issue
Run a windows nightly job.

#### Expected behavior
All tests are run and job ends in a timely manner.

#### Actual behavior
Sometimes the job hangs, and never closes until someone manually ends it.

#### Additional information
This is the job output:

<details>

```
Starting >>> test_launch_ros
06:57:09 ============================= test session starts =============================
06:57:21 platform win32 -- Python 3.8.3, pytest-6.2.5, py-1.10.0, pluggy-1.0.0
06:57:21 cachedir: C:\ci\ws\build\test_launch_ros\.pytest_cache
06:57:21 rootdir: C:\ci\ws\src\ros2\launch_ros\test_launch_ros, configfile: pytest.ini
06:57:21 plugins: ament-copyright-0.11.2, ament-flake8-0.11.2, ament-lint-0.11.2, ament-mypy-0.11.2, ament-pep257-0.11.2, ament-xmllint-0.11.2, launch-testing-0.19.0, launch-testing-ros-0.15.0, colcon-core-0.6.1, cov-3.0.0, mock-3.6.1, repeat-0.9.1, rerunfailures-10.2
06:57:21 collected 121 items
06:57:21 
06:57:21 test\test_copyright.py .                                                 [  0%]
06:57:24 test\test_flake8.py .                                                    [  1%]
06:57:25 test\test_pep257.py .                                                    [  2%]
06:57:25 test\test_launch_ros\actions\test_composable_node_container.py ...       [  4%]
06:57:42 test\test_launch_ros\actions\test_lifecycle_node.py ..                   [  6%]
06:57:42 Build was aborted
```
</details>

It has happened in the buildfarm at least a couple of times:

https://ci.ros2.org/job/nightly_win_deb/2151/
https://ci.ros2.org/view/nightly/job/nightly_win_rel/2101/

It has happened in some `ci_windows` jobs used to test https://github.com/ros2/rclcpp/pull/1814.

http://ci.ros2.org/job/ci_windows/15778/
https://ci.ros2.org/job/ci_windows/15785/

Blast545, 2021-11-30:
It happened again here: https://ci.ros2.org/view/nightly/job/nightly_win_rep/2443 (over the weekend)

hidmic, 2021-12-08:
@Blast545 is this is still an issue?

Blast545, 2021-12-14:
I haven't seen it happening anytime recently, I'll close it for the time being and re open it if the same error happens again.

closed 2021-12-14 by Blast545