ros2/launch_ros #286: Launch delay argument for nodes or other launch files

pepisg, 2021-11-16:
## Feature request

#### Feature description
Hi! it would be great if an argument could be added to add a delay for launching a given node or launch file.
I think this could be useful to control the CPU load of the computing platform when launching a relatively big number of nodes (~40) by preventing all of them from launching at the same time. 

If this feature already exits or can be easily implemented please point me in the right direction.

Thanks in advance for your help!

joncppl, 2021-12-01:
in python launch files I implemented this by wrapping each node to delay with a launch.action.TimerAction.

It would be nicer to just be able to add the delay as an argument argument though! 

eg. launch_ros.actions.Node could emit a TimerAction wrapping what it does now

jacobperron, 2022-01-06:
Yeah, I'd rather see examples of people using `TimerAction` to accomplish this task and we can later consider adding a utility (or simply an example) to launch as common patterns arise.