ros2/launch_ros #291: Provide more meaningful output on error

SteveMacenski, 2021-11-29:
## Feature request

#### Feature description

When launch cannot find a package, a launch file does not exist, a node doesn't exist, or generally speaking something goes wrong, the traceback provided really doesn't help in debugging the issue. I would really like these to be caught and categorized so that users receive a meaningful error message back about what they're doing wrong to be able to fix, ideally with the offending line.  

I'm finding it quite difficult when building new launch files and errors occur to figure out what's wrong. The terminal gets filled with internal launch tracebacks that infrequently help in diagnosing the underlying issue. Especially for common errors, we should have common error messages that tells people why it failed to execute and suppress the traceback. Surely this is an uphill battle to cover all situations, but if we simply cover the top 10, I think that would be a huge quality of life improvement

- Package doesn't exist on system
- Launch file or node doesn't exist on system
- Python syntax errors
- Field given to a `launch` API is invalid

SteveMacenski, 2022-05-24:
@hidmic I know this is probably low priority, but is this on anyone's radar for an upcoming release. This is a big quality of life issue in trying to debug modest robotic systems

jacobperron, 2022-05-31:
I'm not aware of anyone planning to tackle this issue, though I'm happy to review proposals if anyone in the community would like to work on it.

IIUC, errors are communicated via exceptions that [launch catches and logs](https://github.com/ros2/launch/blob/074cd2903ddccd61bce8f40a0f58da0b7c200481/launch/launch/launch_service.py#L344-L347). I guess there are situations where the errors are not being caught and we're getting undesired tracebacks.

For instance, if I launch a node action for a package that doesn't exist, the message looks _okay_ to me:

```
jacob@warner ~/tmp> ros2 launch talker.launch.py        
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2022-05-31-14-41-11-720542-myhost-19000
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'foo' not found, searching: ['/opt/ros/rolling']"
```
Of course it  would be better if it gave us a file and line number, but that might be something that needs to be added upstream in launch (I'm not sure). If there are particular messages that are cryptic, or a category of error that is causing a traceback, we can probably improve those on a case-by-case basis.

SteveMacenski, 2022-06-01:
An example is syntactical errors, they result in not super helpful things like below, which in no place give you the line number of the offending object

```
[INFO] [launch]: All log files can be found below /home/XYZ/.ros/log/2022-05-24-15-03-11-094353-tb07-286055
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("'DeclareLaunchArgument' object is not iterable")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 159, in execute
    set_launch_configuration_actions.append(SetLaunchConfiguration(name, value))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/set_launch_configuration.py", line 49, in __init__
    self.__value = normalize_to_list_of_substitutions(value)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/normalize_to_list_of_substitutions_impl.py", line 44, in normalize_to_list_of_substitutions
    return [normalize(y) for y in cast(Iterable, subs)]
TypeError: 'DeclareLaunchArgument' object is not iterable
```

jacobperron, 2022-06-03:
Identifying line numbers related to syntax issues is probably best as a feature request in the launch repo; certainly a feature worth adding :+1:

jacobperron, 2022-06-03:
Here's a relevant ticket: https://github.com/ros2/launch/issues/412
I suppose it extends to Python launch files as well.

rolker, 2025-02-05:
I encountered this same problem and submitted this PR which helps. https://github.com/ros2/launch/pull/823