ros2/launch_ros #292: WaitForTopics Utility fails on WIndows Debug builds

adityapande-1995, 2021-12-07:
## Bug report

- Operating System:
  - Windows 10 
- Installation type:
  - CI installation, with debug builds
- Version or commit hash:
  - Came up here : https://github.com/ros2/launch_ros/pull/288
- Client library (if applicable):
  - NA

#### Steps to reproduce issue
Windows debug builds (in CI) fail for the ``WaitForTopics`` utility, where the test cases cannot receive messages on a topic.
Link to failing CI builds : https://ci.ros2.org/job/ci_windows/15864/

#### Expected behavior
Ideally messages should be received on topics specified, as the do for normal Windows builds: https://ci.ros2.org/job/ci_windows/15868/

#### Actual behavior
Tests fail on windows debug builds with the message : 
```
FAIL: wait_for_topic_launch_test.TestFixture.test_topics_successful
----------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\ci\ws\src\ros2\launch_ros\launch_testing_ros\test\examples\wait_for_topic_launch_test.py", line 58, in test_topics_successful
    with WaitForTopics(topic_list, timeout=2.0) as wait_for_node_object_1:
  File "C:\ci\ws\src\ros2\launch_ros\launch_testing_ros\launch_testing_ros\wait_for_topics.py", line 96, in __enter__
    raise RuntimeError('Did not receive messages on these topics: ',
RuntimeError: ('Did not receive messages on these topics: ', {'chatter_0', 'chatter_2', 'chatter_1', 'chatter_3', 'chatter_4'})
```
The test times out and fails.
----

audrow, 2022-03-29:
This seems to have come up on the [`nightly_linux_debug`](https://ci.ros2.org/job/nightly_linux_debug/2251/testReport/junit/launch_testing_ros.test.examples/wait_for_topic_launch_test/wait_for_topic_launch_test/).

Blast545, 2022-03-29:
@audrow I think it's different. The previous error on Windows was a test regression. In this case the errors are failing in the same place, but it seems the build is red because there's a failure to close to the tests successfully:

```
[python-20] Traceback (most recent call last):^M
[python-20]   File "/home/jenkins-agent/workspace/nightly_linux_debug/ws/src/ros2/launch_ros/launch_testing_ros/test/examples/talker.py", line 51, in <module>^M
[python-20]     main()^M
[python-20]   File "/home/jenkins-agent/workspace/nightly_linux_debug/ws/src/ros2/launch_ros/launch_testing_ros/test/examples/talker.py", line 47, in main^M
[python-20]     rclpy.shutdown()^M
[python-20]   File "/home/jenkins-agent/workspace/nightly_linux_debug/ws/install/rclpy/lib/python3.10/site-packages/rclpy/__init__.py", line 126, in shutdown^M
[python-20]     _shutdown(context=context)^M
[python-20]   File "/home/jenkins-agent/workspace/nightly_linux_debug/ws/install/rclpy/lib/python3.10/site-packages/rclpy/utilities.py", line 58, in shutdown^M
[python-20]     return context.shutdown()^M
[python-20]   File "/home/jenkins-agent/workspace/nightly_linux_debug/ws/install/rclpy/lib/python3.10/site-packages/rclpy/context.py", line 102, in shutdown^M
[python-20]     self.__context.shutdown()^M
[python-20] rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at /home/jenkins-agent/workspace/nightly_linux_debug/ws/src/ros2/rcl/rcl/src/rcl/i\
nit.c:241^M
```

adityapande-1995, 2022-03-29:
The tests have been disabled on Windows : https://github.com/ros2/launch_ros/blob/4192b02af65cb3a8693cc644952f66b71bfbaa8b/launch_testing_ros/test/examples/wait_for_topic_launch_test.py#L52

Blast545, 2022-03-29:
@adityapande-1995 Yeah, you are right. This is currently affecting `nightly_linux_debug` jobs:
https://ci.ros2.org/view/nightly/job/nightly_linux_debug/2251/

But I'll open a new thread for it. As this test is disabled on Windows I think you can close this issue.

adityapande-1995, 2022-03-29:
Thanks @Blast545 !

closed 2022-03-29 by adityapande-1995

Ryanf55, 2024-02-20:
> @adityapande-1995 Yeah, you are right. This is currently affecting `nightly_linux_debug` jobs: https://ci.ros2.org/view/nightly/job/nightly_linux_debug/2251/
> 
> But I'll open a new thread for it. As this test is disabled on Windows I think you can close this issue.

Where's the new thread? I want to see if this is closed, or is still an issue.