ros2/launch_ros #294: Make ComposableNode support "condition"

doisyg, 2022-01-26:
## Feature request
Make `ComposableNode` support a `condition`, for instance:

```
              ComposableNode(
                  package = 'my_package',
                  plugin = 'my_package::my_plugin',
                  name = 'my_plugin_node',
                  condition = UnlessCondition(use_sim_time),
               ),
```
#### Feature description

Same feature as for `Node`, `IncludeLaunchDescription` and `ComposableNodeContainer`

adityapande-1995, 2022-04-14:
Just tested this locally, and ``condition`` works for ``Node`` : 
```
import launch
import launch.actions
from launch.conditions import IfCondition
from launch.substitutions import LaunchConfiguration
import launch_ros.actions

def generate_launch_description():
    return launch.LaunchDescription([
        launch.actions.DeclareLaunchArgument(name='flag', default_value='False'),

        launch_ros.actions.Node(
            executable='talker',
            package='demo_nodes_cpp',
            name='demo_node',
            condition=IfCondition(LaunchConfiguration('flag'))
        ),
    ])
```

closed 2022-04-27 by adityapande-1995

sea-bass, 2023-06-10:
+1 to @alexMarFar -- we just had a MoveIt user run into this issue because we had an `UnlessCondition` in a new composable node, and it only works on Rolling.

vmayoral, 2023-06-16:
+1 in here. Can we please get this into Humble?

mcres, 2023-06-16:
I'm also running into this issue, it would be very nice to have this feature in Humble.

sea-bass, 2023-06-16:
FYI they did kick off a backport a few days ago: https://github.com/ros2/launch_ros/pull/363

Thanks, @clalancette !